open62541/open62541 #6003: unable to receive events after update

PvanIeperen, 2023-09-08:
## Description
At the moment I work on a client that must connect to a server that is running on a robot form another team in our company. This team is writing the server in C++ with the open62541 library. And we are using the .Net library from OpcFoundation.NetStandard 1.4.371.96.

The robot team has updated their library to the newest version because they need it for the HMI of the robot system.
After this update the events are not working anymore. Before the update everything was working.
open62541 v1.2.2 (events are working)
open62541 v1.3.6 (not working)

I need to figure out how I can change the client in such a way that I can receive the events again.

Is someone familiar with this issue? Can someone think of some possible hints that are use full?
are there any changed made in the event system? 

regards Patrick

## Background Information / Reproduction Steps
UAExpert still receives the events from the server after the update.

If I create a plc (B&R) that fire opcua events than our software is still working.

our .Net client is not working anymore with the updated version of the open62541 library.

## Checklist

Please provide the following information:

 - [ X] open62541 Version (release number or git tag):1.3.6
 - [ X] Other OPC UA SDKs used (client or server): client OpcFoundation.NetStandard 1.4.371.96. and UAexpert
 - [ X] Operating system: Linux and Windows
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

maucm92, 2023-09-08:
Hi!
Can you provide the logs of the client and server? What node are you using for receiving events?
I detected that building a OPC UA Server with open62541 v1.3.6 (also 1.3.7) and NS0 = reduced produce this behaivor:
* Prosys OPC UA Browser can't suscribe to events on server node: UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER)
* Open62541 OPC UA Client also can't suscribe to events on server node

The error ocurrs on monitored item creation, fails with message BadNodeIdInvalid. 

OPC Foundation client can suscribe to node events.
  
Building with NS0 = full fix the error only in Prosys OPC UA Browser (not in Open62541 OPC UA Client)

maucm92, 2023-09-08:
> Building with NS0 = full fix the error only in Prosys OPC UA Browser (not in Open62541 OPC UA Client)

**Update:** The error in Open62541 OPC UA Client apparently it's mine. 

I tried to simplify the source codes client_eventfilter.c and server_random_events.c and it seems that I made a mistake. 
Using the files without modifications the code just works. Now I can receive events. 

In conclusion **NS0 = full** completely fixes the error for v1.3.6

jpfr, 2023-09-09:
@PvanIeperen we want to support all vendors on the other end.
If you can share (also privately) a wireshark network dump, that makes our life easier.
Or detailed log messages if you can access those.

PvanIeperen, 2023-09-11:
> **Update:** The error in Open62541 OPC UA Client apparently it's mine.
>
> I tried to simplify the source codes client_eventfilter.c and server_random_events.c and it seems that I made a mistake. Using [...]

@maucm92 

Hi Mauro, 
We have the build option set to FULL. This is still not solve our issue.
The issue that we are facing is only with events. With monitored items we do not have any issues.

regards Patrick

jpfr, 2023-09-14:
Yeah. So the network dump looks perfectly fine. As you said it uses UAExpert.

I tried the Softing Client that also uses .net Standard.
Events are received just fine.

[image]

Do you have a (full) network trace for the case where it doesn't work?
You can also send over a private channel if this is sensitive...

Otherwise you might have access to more detailed log information.

jpfr, 2023-09-18:
I see traffic. But only the publish requests/responses.
There is not much there. Just that no events come through (surprise).

There could be an error communicated when the Event MonitoredItem is created initially.
But that needs to be included in the network dump.

Anything interesting in the log messages?

PvanIeperen, 2023-09-19:
Something is going wrong during the creation of a monitored item.

[image]

But that is strange we use on the client side the same nodeId to subscribe as with the previous version of the open62541 lib.

I will attach the wireshark dump when the client is starting up.

[monitorItemCreation.zip](https://github.com/open62541/open62541/files/12662269/monitorItemCreation.zip)

jpfr, 2023-09-21:
It appeas the event filter is not valid.
And we have added checks in 1.3 that were not present before.

You have added a filter "oftype" with target `ns=2;i=263`.
This is a custom definition in namespace 2.
But the target does not inherit from `BaseEventType`.
That is not allowed.

PR #6015 adds the functionality to return detailed status information for the filters.
So this becomes easier to debug.

In addition, if you cannot change the underlying information model, we could add a server config flag to demote the error to a warning.

PvanIeperen, 2023-09-21:
Hi Julius,
What do you mean by " _the target does not inherit from BaseEventType._"?
If we check the server then we see that our event derives from BaseEventType. See screenshot below.

[image]

We have changed the filter now on the .Net Client side to the base event (i=2041). Is that what you mend? Or did we didn't understand you correctly? 
But we are not able to receive the event. 
```
  WhereClause: ContentFilter
                                    Elements: Array of ContentFilterElement
                                        ArraySize: 1
                                        [0]: ContentFilterElement
                                            FilterOperator: OfType (0x0000000e)
                                            FilterOperands: Array of ExtensionObject
                                                ArraySize: 1
                                                [0]: ExtensionObject
                                                    TypeId: ExpandedNodeId
                                                        EncodingMask: 0x01, EncodingMask: Four byte encoded Numeric
                                                            .... 0001 = EncodingMask: Four byte encoded Numeric (0x1)
                                                            .0.. .... = has server index: False
                                                            0... .... = has namespace uri: False
                                                        Namespace Index: 0
                                                        Identifier Numeric: 597
                                                    EncodingMask: 0x01, has binary body
                                                        .... ...1 = has binary body: True
                                                        .... ..0. = has xml body: False
                                                    LiteralOperand: LiteralOperand
                                                        Value: Variant
                                                            Variant Type: NodeId (0x11)
                                                            Value: NodeId
                                                                .... 0001 = EncodingMask: Four byte encoded Numeric (0x1)
                                                                Namespace Index: 0
                                                                Identifier Numeric: 2041
```

[capture request after filter changed.txt](https://github.com/open62541/open62541/files/12682801/capture.request.after.filter.changed.txt)

jpfr, 2023-09-21:
The only place where the StatusCode `BadNodeIdInvalid` is set is here:
https://github.com/open62541/open62541/blob/64b79a6fe62d50b30cd71e1ee48b5f6f40a71314/src/server/ua_subscription_events_filter.c#L1290
That is where we check whether the EventType is valid.

We will build on top of #6015 to try to reproduce.

PvanIeperen, 2023-09-21:
Is there something we can do? 
We control both the server (Open62541 ) and the client (UA-.NETStandard-1.5.372.1-rc) part.

jpfr, 2023-09-22:
The 1.3 branch has received an update.
Now detailed error information is returned when the filter validation fails.

Using that will give further insight.

Otherwise, if you compile the server yourself, do you have a chance to debug the place I indicated or add printf statements?

PvanIeperen, 2023-10-03:
Hi Julius,
We changed the SelectClause in our client and that solved the problem.

Really thanks for all your help.
Regards Patrick

jpfr, 2023-10-03:
you're welcome

closed 2023-10-03 by jpfr