open62541/open62541 #6005: Functions declared as deprecated in 1.2.7 comapred to 1.1.6

gitlae, 2023-09-09:
## Description
Have been using 1.1.x for a while but recently decided to head for the latest 1.3.x version, taking the first step with 1.2.x.

When compiling 1.2.7 on Windows I get the following warning about deprecated functions:
```
opcua.cpp(1162): warning C4996: 'UA_BrowseRequest_deleteMembers': was declared deprecated ...
opcua.cpp(1162): warning C4996: 'UA_BrowseResponse_deleteMembers': was declared deprecated ...
opcua.cpp(1162): warning C4996: 'UA_WriteRequest_deleteMembers': was declared deprecated ...
opcua.cpp(1162): warning C4996: 'UA_WriteResponse_deleteMembers': was declared deprecated ...
```
I have search for information about how to handle the deprecated status but not found any information.

My question is what is the correct action to take:
1. Do I just removed the call to the deprecated functions ?
2. Do I have to refactor the code to use non deprecated functions. If so, is there any information available on that topic?

I see the same messages using 1.3.6

## Background Information / Reproduction Steps
-
## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-09-09:
Use _clear instead.
The methods were just renamed to be more … ahem … clear.

closed 2023-09-09 by jpfr

gitlae, 2023-09-10:
Thanks for the prompt reply.