open62541/open62541 #6010: Release 1.3.7 declares bad version in CMakeLists.txt

herrhotzenplotz, 2023-09-11:
## Description

## Background Information / Reproduction Steps

As the maintainer of the FreeBSD Port (https://www.freshports.org/devel/open62541) I updated to the new release.

The initial rebuild with the package builder failed due to the CMakeLists.txt in the version 1.3.7 tag declaring the library version as 1.3.6 and thus producing a incorrectly named shared library with a suffix `.so.1.3.6`. This caused the package builder to fail:

https://poudriere.herrhotzenplotz.de/build.html?mastername=release-latest-main&build=2023-09-11_12h41m39s

I patched the CMakeLists.txt locally to work around this issue. I do not know whether this is worth fixing however I thought that it is worth reporting.

See also: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=273718

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.3.7
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: FreeBSD 13.2-RELEASE-p3
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-09-14:
Good point. This will be fixed later today.
But we need a new minor tag — we don’t change tags once they are defined.

jpfr, 2023-10-09:
The new v1.3.8 release is correctly tagged in the CMakeLists.txt.
Reopen this if problems remain.

closed 2023-10-09 by jpfr

herrhotzenplotz, 2025-04-16:
@jpfr Just a heads-up, this issue happened again in v1.4.11 which declares itself as v1.4.10.
If there is a release checklist it'd be nice if there is an entry in it to double check this 😄