open62541/open62541 #6019: Multicast Discovery causes continuously increasing network traffic

r-sin, 2023-09-22:
## Description
Investigating the needed resources for using Discovery/mDNS in a product, WE have observed the following behavior.

After the usual mDNS startup traffic, the periodic mDNS updates are getting longer and larger on each cycle. With only one LDS-ME running, after about 4-5 hours this leads to a constant traffic :
```bash
                                                                     | | | | | | | | | | | | | | | | | | | |
                                                                                     ...
                                                                     | | | | | | | | | | | | | | | | | | | |
                                                       | | | |       | | | | | | | | | | | | | | | | | | | |
                                             | | |     | | | |       | | | | | | | | | | | | | | | | | | | |
                                 | |         | | |     | | | |       | | | | | | | | | | | | | | | | | | | |
                     |           | |         | | |     | | | |       | | | | | | | | | | | | | | | | | | | |
    |||||___........_||__________||||________||||||____|||||||| .... |||||||||||||||||||||||||||||||||||||||
                    +5min       +5min       +5min       +5min       after 4-5 hours with one LDS-ME running
LDS-ME startup traffic   PTR RR update every 5 minutes: message size and number of messages are growing 
```

This behavior is independent of
- whether other Servers have registered with the LDS-ME or not
- the platform: observed on Windows (17 answer records in the same query) and Linux (33 records)
- open62541 version (observed in 1.3.4 and 1.3.6)

You can start one single LDS-ME (from the discovery examples) to observe this. No other servers have to register with the LDS-ME.

After 50% of the record lifetime (-> 5min) there comes a PTR query with n responses in the same massage. n is getting larger each time (until 17 on Windows, 33 on Linux), so these messages are getting larger).
After each of these queries there comes a "normal" PTR query with only one response. The requested PTR RR is always the same : _opcua-tcp._tcp.local: type PTR, class IN, "QM" question.
In the normal query, the TTL is set back to 10 minutes each time. In the query with the increasing number of records, the TTL is decreasing in the particular answers from about 5 minutes.

In the first update after 5 minutes. 6 of these pairs [1 Query with many answers + 1 query with only one answer] are sent out. Every 5 minutes, 6 more pairs are added. So, the number of messages is getting larger every 5 minutes.
Finally, after 4 - 5 hours, each update cycle takes longer than 5 minutes, so the cycles overlap and there is a constant traffic on the network.

## Background Information / Reproduction Steps
Checkout v1.3.4 or v1.3.6
Enable Discovery + Multicast in CMake and build
Start the discovery example LDS-ME, watch the mDNS protocol messages in Wireshark

## Checklist

Please provide the following information:

 - [ ] open62541 Version (tried v1.3.4 and v1.3.6: 
 - [ ] Operating system: Windows 10, Ubuntu

jpfr, 2023-09-28:
Thanks for the detailed report.

we have revamped the control flow for multicast in the 1.4 series.

is it a possibility for you to switch to the 1.4 branch? We can debug further into 1.3, but much was improved for 1.4 already.

r-sin, 2023-09-29:
Yes, we can consider 1.4 - thanks,

o6-nbeier, 2026-10-22:
Closing for no activity.

closed 2026-10-22 by o6-nbeier