open62541/open62541 #6031: Unable to connect to server_ctt example using Basic128Rsa15

mister-potter, 2023-09-29:
## Description

## Background Information / Reproduction Steps

Using v1.3.7 branch, I have built server_ctt.c example. From the self signing certificate script, I've generated certificate and key for my server and client. Signed the key using SHA1 algorithm and loaded the same in UaExpert client.

Upon connecting using security policy Basic128Rsa15, I get the following error.

> [2023-09-29 07:50:26.583 (UTC+0550)] warn/session       SecureChannel 2 | Session "urn:localhost:UnifiedAutomation:UaExpert" | ActivateSession: User token signature check failed with StatusCode BadSecurityChecksFailed

```
[2023-09-29 07:50:13.953 (UTC+0550)] warn/server        AccessControl: Unconfigured AccessControl. Users have all permissions.
[2023-09-29 07:50:13.953 (UTC+0550)] info/server        AccessControl: Anonymous login is enabled
[2023-09-29 07:50:13.956 (UTC+0550)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=integer): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2023-09-29 07:50:13.956 (UTC+0550)] info/session       SecureChannel 0 | Session "Administrator" | AddNode (ns=1;s=uinteger): The value is empty. But this is only allowed for BaseDataType. Create a matching default value.
[2023-09-29 07:50:13.968 (UTC+0550)] info/network       TCP network layer listening on opc.tcp://localhost:4850/
[2023-09-29 07:50:21.912 (UTC+0550)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2023-09-29 07:50:21.914 (UTC+0550)] info/channel       Connection 5 | SecureChannel 1 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 300.00s
[2023-09-29 07:50:21.916 (UTC+0550)] info/channel       Connection 5 | SecureChannel 1 | CloseSecureChannel
[2023-09-29 07:50:21.916 (UTC+0550)] info/network       Connection 5 | Closed
[2023-09-29 07:50:25.308 (UTC+0550)] info/network       Connection 5 | New connection over TCP from 127.0.0.1
[2023-09-29 07:50:25.313 (UTC+0550)] warn/server        PKI plugin unconfigured. Accepting the certificate.
[2023-09-29 07:50:25.327 (UTC+0550)] info/channel       Connection 5 | SecureChannel 2 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15 and a revised lifetime of 300.00s
[2023-09-29 07:50:25.345 (UTC+0550)] info/session       SecureChannel 2 | Session "urn:localhost:UnifiedAutomation:UaExpert" | Session created
[2023-09-29 07:50:26.583 (UTC+0550)] warn/session       SecureChannel 2 | Session "urn:localhost:UnifiedAutomation:UaExpert" | ActivateSession: User token signature check failed with StatusCode BadSecurityChecksFailed
[2023-09-29 07:50:26.660 (UTC+0550)] info/session       SecureChannel 2 | Session "urn:localhost:UnifiedAutomation:UaExpert" | Closing the Session
[2023-09-29 07:50:26.661 (UTC+0550)] info/channel       Connection 5 | SecureChannel 2 | CloseSecureChannel
[2023-09-29 07:50:26.661 (UTC+0550)] info/network       Connection 5 | Closed
```
If my approach towards generating certificate and key is wrong, please guide me on how to create it for Basic128Rsa15 policy.

Used CMake options:

```bash
cmake -DUA_BUILD_EXAMPLES=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v.1.3.7
 - [x] Other OPC UA SDKs used (client or server): UaExpert
 - [x] Operating system: Ubuntu 22.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-10-04:
You use the certificate for session authentication or (only) for the encrypted SecureChannel?

mister-potter, 2023-10-04:
For session authentication

> You use the certificate for session authentication or (only) for the encrypted SecureChannel?

jpfr, 2023-10-04:
On 1.4/master branch there are two "CertificateVerification" plugin instances.
One for the SecureChannel and another for the session.
Make sure the certificate is trusted in both.

mister-potter, 2023-10-04:
Thanks for the response.
I'm currently using v1.3.7 release. 

Is the expectation same in that branch as well?

jpfr, 2023-10-04:
No.
But you still need to add the certificate (or the issuer in a certificate chain) to the trustlist.
See the server_ctt example.

mister-potter, 2023-10-08:
I tried following your suggestion and added the certificate in Trust List store.
Still the result was same. Hence, tried to dig deeper and figured out the following.

For the Basic128Rsa15 policy, key must be having strength either 1024 or 2048 bits. And the signing algorithm must be SHA1 with RSA encryption. I have created the certificate and key in same manner.

Upon establishing a new connection, SecureChannel is requested with same policy. But in the file `ua_services_session.c` file, the following function [Service_ActivateSession()](https://github.com/open62541/open62541/blob/pack/v1.3.7/src/server/ua_services_session.c#L784) -> `request->userTokenSignature.algorithm` is pointing to rsa-sha256 algorithm.

Upon execution, user token signature verification fails. Could you please clarify why signature is referring to rsa-sha256 instead of sha1?

NoelGraf, 2023-10-09:
You need to adjust the policy accordingly in the AccessControl.
`
UA_String policyUri = UA_STRING("http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15");
UA_AccessControl_default(config, true, &policyUri, usernamePasswordsSize, usernamePasswords);
`

mister-potter, 2023-10-09:
Thank you @NoelGraf .
With the above mentioned solution, I'm able to configure the Basic128Rsa15 policy and establish a connection as well.

closed 2023-10-09 by mister-potter