open62541/open62541 #6040: Deadlock calling UA_Server_addPubSubConnection from low priority task

keba-estr, 2023-10-04:
If UA_Server_addPubSubConnection is called in a linux system from a task with lower priority than the server task that calls UA_Server_run, a deadlock occurs.

## Description

## Background Information / Reproduction Steps

The function UA_Server_run waits for a event with UA_EventLoopPOSIX_pollFDs (epoll_wait) with the locked event loop mutex, while the function UA_Server_addPubSubConnection calls UA_PubSubConnection_connectUDP and UDP_openConnection which tries to lock the event loop mutex.

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ X ] open62541 Version (release number or git tag): 1.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

keba-estr, 2023-10-04:
Is it possible/safe to call UA_EventLoopPOSIX_pollFDs with unlocked eventloop mutex?

keba-estr, 2023-10-04:
Priority inheritance does not solve the problem, because only the priority of the task that locks the mutex is increased to a task waiting for the lock, not vice verse.

keba-estr, 2023-10-05:
@jpfr: Reguarding your latest refactoring PRs, did you already face this problem that the event loop waiting condition is done in locked state?
Is there a safe point where the lock can be opened before waiting inside UA_EventLoopPOSIX_pollFDs (or the 'select' use case)?

keba-estr, 2023-10-05:
For the 'select' use case, the call of UA_select without locked EventLoop mutex should suffice to resolve the priority problem, as all prameters of UA_select are local variables. Is it possible that the EventLoop itself gets invalid after calling UA_select (i.e. when locking again)?

keba-estr, 2023-10-05:
For the 'epoll' use case, the epollfd must be stored in a local variable to call epoll_wait without locked mutex. Save question as above about invalid EventLoop after call raises.

jpfr, 2023-10-05:
Good find.
I guess we need to do give up the `elMutex` before going into the application-callback.
That will be a bit of work when multiple fds were triggered in the same select.
We have to check for each whether the connection was closed in the previous application-callback.

As a "quick fix" you can just release the `elMutex` around the application-callback.
The actual fix will land in a few days.

keba-estr, 2023-10-05:
@jpfr: Which application-callback are you referring to?

keba-estr, 2023-10-05:
I created PR  #6042 with the unlocked eventloop mutex before the waiting condition (without additional checks about changed connections). This can be used as a starting point for improvements.

keba-estr, 2023-11-03:
Resolved with PR #6042

closed 2023-11-03 by keba-estr