open62541/open62541 #6051: New client connection logic returns BadServiceUnsupported when the discoveryUrl and the endpointUrl are different

simonyg, 2023-10-09:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=OFF -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_ENABLE_ENCRYPTION=ON -DUA_ENABLE_ENCRYPTION_OPENSSL=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON -DUA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS=ON -DUA_NAMESPACE_ZERO=FULL -DUA_BUILD_EXAMPLES=ON ..

```
New client connection logic does not work when discoveryUrl and the endpointUrl are different.

Perhaps, this is the fix in ua_client_connect.c line #793:

    /* Close the SecureChannel if a different SecurityPolicy is defined by the Endpoint */
    /* reset discoveryURL so the next HEL message uses the endpointUrl
    if(!UA_String_equal(&client->discoveryUrl, &client->endpointUrl) ||
       client->config.endpoint.securityMode != client->channel.securityMode ||
       !UA_String_equal(&client->config.endpoint.securityPolicyUri,
                        &client->channel.securityPolicy->policyUri)) {
        UA_String_clear(&client->discoveryUrl);
        UA_String_copy(&client->endpointUrl, &client->discoveryUrl);
        closeSecureChannel(client);
    }

Steps to reproduce:

Example client_connect is used to work @v1.3.4 for the demo OPCUA server opc.tcp://milo.digitalpetri.com:62541/milo
```
open62541/build/bin/examples/client_connect opc.tcp://milo.digitalpetri.com:62541/milo

got fallback qt version 0x50200[2023-10-09 11:44:47.052 (UTC-0700)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-10-09 11:44:49.286 (UTC-0700)] info/channel	Connection 5 | SecureChannel 1439807 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-09 11:44:49.286 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-10-09 11:44:49.311 (UTC-0700)] info/client	Selected endpoint 0 in URL opc.tcp://milo.digitalpetri.com:62541/milo with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-10-09 11:44:49.311 (UTC-0700)] info/client	Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-10-09 11:44:49.345 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-10-09 11:44:49.367 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-10-09 11:44:49.368 (UTC-0700)] info/userland	Connected!
[2023-10-09 11:44:49.391 (UTC-0700)] info/userland	The server date is: 09-10-2023 18:44:49.391
[2023-10-09 11:44:49.416 (UTC-0700)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
```
It is broken @v1.3.7
```
open62541/build/bin/examples/client_connect opc.tcp://milo.digitalpetri.com:62541/milo

got fallback qt version 0x50200[2023-10-09 11:48:43.191 (UTC-0700)] warn/userland	AcceptAll Certificate Verification. Any remote certificate will be accepted.
[2023-10-09 11:48:45.426 (UTC-0700)] info/channel	Connection 5 | SecureChannel 1439885 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-09 11:48:45.426 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-10-09 11:48:45.447 (UTC-0700)] info/client	Use the EndpointURL opc.tcp://milo.digitalpetri.com:62541/milo/discovery returned from FindServers
[2023-10-09 11:48:45.447 (UTC-0700)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: Good
[2023-10-09 11:48:45.509 (UTC-0700)] info/channel	Connection 5 | SecureChannel 1439886 | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-09 11:48:45.509 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-10-09 11:48:45.536 (UTC-0700)] warn/client	The server returned Endpoints with a different EndpointUrl opc.tcp://milo.digitalpetri.com:62541/milo/discovery than was used to initialize the connection: opc.tcp://milo.digitalpetri.com:62541/milo/discovery. Some servers require a complete match of the EndpointUrl/DiscoveryUrl (including the path) to return all endpoints.
[2023-10-09 11:48:45.536 (UTC-0700)] info/client	Selected endpoint 0 in URL opc.tcp://milo.digitalpetri.com:62541/milo with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-10-09 11:48:45.536 (UTC-0700)] info/client	Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-10-09 11:48:45.558 (UTC-0700)] info/client	Received a ServiceFault response
[2023-10-09 11:48:45.558 (UTC-0700)] info/client	The ServiceResult has the StatusCode BadServiceUnsupported
[2023-10-09 11:48:45.559 (UTC-0700)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: BadServiceUnsupported
[2023-10-09 11:48:45.559 (UTC-0700)] info/userland	Could not connect
[2023-10-09 11:48:45.559 (UTC-0700)] info/client	Client Status: ChannelState: Closed, SessionState: Closed, ConnectStatus: BadServiceUnsupported
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.7
 - [x] Other OPC UA SDKs used (client or server): client
 - [x] Operating system: Any
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

xydan83, 2023-12-04:
I have the same situation. The server has different  discoveryUrl and the endpointUrl. 
[image]

When I try to connect with version 1.3.4, everything is ok.
[image]

But in 1.3.5 I have a trouble
[image]

But it helps me to hide a couple of lines in **ua_client_connect.c**.
[image]

I don't know why this is done: "**hello.endpointUrl = client->discoveryUrl;**".

I'm trying to understand...

jpfr, 2023-12-12:
This change was added for GetEndpoints on servers that require the exact URL returned by the FindServers service.

The behavior for reconnecting with different EndpointUrl was further improved over the last days on the 1.4 branch.
Is it possible that you try with the current 1.4 branch?

Can you tell us which server/SDK is used on the other end?
So we can include that in manual tests we do before a release.
You can also tell us on a private channel if the information is sensitive.

simonyg, 2023-12-12:
Using the latest 1.4 branch, this seems to be working. See logs below. Thank you!

The OPC UA server is from https://github.com/eclipse/milo which is another open-source implementation of OPC UA. Its current target OPC UA version is 1.03. We use it mainly for testing since they keep this demo OPC UA Server running all the time. 

```
build/bin/examples/client_connect opc.tcp://milo.digitalpetri.com:62541/milo
[2023-12-12 07:49:54.791 (UTC-0800)] info/eventloop	Starting the EventLoop
[2023-12-12 07:49:54.791 (UTC-0800)] warn/client	skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-12-12 07:49:55.089 (UTC-0800)] info/network	TCP 5	| New connection to "milo.digitalpetri.com" on port 62541
[2023-12-12 07:49:55.166 (UTC-0800)] info/channel	TCP 5	| SC 2820184	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-12-12 07:49:55.168 (UTC-0800)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-12-12 07:49:55.184 (UTC-0800)] info/client	Use the EndpointURL opc.tcp://milo.digitalpetri.com:62541/milo/discovery returned from FindServers
[2023-12-12 07:49:55.184 (UTC-0800)] info/channel	TCP 5	| SC 2820184	| SecureChannel closed
[2023-12-12 07:49:55.184 (UTC-0800)] warn/client	skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-12-12 07:49:55.186 (UTC-0800)] info/network	TCP 6	| New connection to "milo.digitalpetri.com" on port 62541
[2023-12-12 07:49:55.186 (UTC-0800)] info/network	TCP 5	| Socket closed
[2023-12-12 07:49:55.261 (UTC-0800)] info/channel	TCP 6	| SC 2820185	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-12-12 07:49:55.261 (UTC-0800)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-12-12 07:49:55.345 (UTC-0800)] warn/client	The server returned Endpoints with a different EndpointUrl opc.tcp://milo.digitalpetri.com:62541/milo/discovery than was used to initialize the connection: opc.tcp://milo.digitalpetri.com:62541/milo/discovery. Some servers require a complete match of the EndpointUrl/DiscoveryUrl (including the path) to return all endpoints.
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Rejecting endpoint 1: security mode doesn't match
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Rejecting endpoint 2: security mode doesn't match
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Rejecting endpoint 3: security mode doesn't match
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Rejecting endpoint 4: security mode doesn't match
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Rejecting endpoint 5: security mode doesn't match
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Rejecting endpoint 6: security mode doesn't match
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Selected endpoint 0 in URL opc.tcp://milo.digitalpetri.com:62541/milo with SecurityMode None and SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-12-12 07:49:55.345 (UTC-0800)] info/client	Selected UserTokenPolicy anonymous with UserTokenType Anonymous and SecurityPolicy 
[2023-12-12 07:49:55.345 (UTC-0800)] info/channel	TCP 6	| SC 2820185	| SecureChannel closed
[2023-12-12 07:49:55.345 (UTC-0800)] warn/client	skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-12-12 07:49:55.347 (UTC-0800)] info/network	TCP 5	| New connection to "milo.digitalpetri.com" on port 62541
[2023-12-12 07:49:55.347 (UTC-0800)] info/network	TCP 6	| Socket closed
[2023-12-12 07:49:55.405 (UTC-0800)] info/channel	TCP 5	| SC 2820186	| SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-12-12 07:49:55.405 (UTC-0800)] info/client	Client Status: ChannelState: Open, SessionState: Closed, ConnectStatus: Good
[2023-12-12 07:49:55.446 (UTC-0800)] info/client	Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-12-12 07:49:55.467 (UTC-0800)] info/client	Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-12-12 07:49:55.467 (UTC-0800)] info/userland	Connected!
[2023-12-12 07:49:55.483 (UTC-0800)] info/userland	The server date is: 12-12-2023 15:49:55.494
[2023-12-12 07:49:55.500 (UTC-0800)] info/channel	TCP 5	| SC 2820186	| SecureChannel closed
[2023-12-12 07:49:55.500 (UTC-0800)] info/client	Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionClosed
[2023-12-12 07:49:55.500 (UTC-0800)] info/network	TCP 5	| Socket closed
[2023-12-12 07:49:55.500 (UTC-0800)] info/eventloop	Stopping the EventLoop
[2023-12-12 07:49:55.500 (UTC-0800)] info/network	UDP	| Shutting down the ConnectionManager
[2023-12-12 07:49:55.500 (UTC-0800)] info/network	TCP	| Shutting down the ConnectionManager
[2023-12-12 07:49:55.500 (UTC-0800)] info/eventloop	The EventLoop has stopped
```

o6-nbeier, 2026-10-22:
Thanks for reporting this.

We reviewed this issue against the currently supported branches. It only affects versions below 1.4, and we can’t reproduce it on 1.4 or later. Versions below 1.4 no longer receive fixes, so we are closing this issue.

If you still see the problem on 1.4 or later, please reopen the issue or open a new one. Include the open62541 version, your platform and build configuration, and a minimal way to reproduce it.

closed 2026-10-22 by o6-nbeier