open62541/open62541 #6057: hello I am trying to build open62541 include my project

bbakcoin, 2023-10-12:
## Description

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): open62541-v1.3.0
 - [ ] Other OPC UA SDKs used (client or server):  simmense plc cim / ua expert
 - [ ] Operating system: window 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

hello guys I am just had few expeirenced c++ 

basically I am working as plc enginner

to me cmake and some cmd tools are really hard to understand

FIRST THING THAT I TRIED BUILD WITH CMAKE

I tried gui and cmd tool following docs of open 62541

**Then open buildopen62541.sln in Visual Studio 2015 and build as usual**

there is no such a file buildopen62541 but I had open62541.sin but all builds are failed

Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	C2220	warning treated as error - no 'object' file generated	open62541-object	c:\users\amorf\open62541-1.3\src\server\ua_subscription.h	1	
Error	LNK1181	cannot open input file 'C:\Users\amorf\open62541-1.3\build\open62541-object.dir\Release\ua_session.obj'	open62541 (open62541\lib\open62541)	C:\Users\amorf\open62541-1.3\build\LINK	1	
Error	LNK1181	cannot open input file '..\bin\Release\open62541.lib'	tutorial_datatypes	C:\Users\amorf\open62541-1.3\build\examples\LINK	1	
Error	LNK1181	cannot open input file '..\bin\Release\open62541.lib'	tutorial_client_firststeps	C:\Users\amorf\open62541-1.3\build\examples\LINK	1	
Error	LNK1181	cannot open input file '..\bin\Release\open62541.lib'	access_control_server	C:\Users\amorf\open62541-1.3\build\examples\LINK	1	
Error	LNK1181	cannot open input file '..\bin\Release\open62541.lib'	tutorial_server_method	C:\Users\amorf\open62541-1.3\build\examples\LINK	1	
[... 28 lines cut ...]
with build cmake and include to my project is and all failed

so I tried to prebuilt pakcage which https://github.com/open62541/open62541/releases

[image]

with that file and I copy and paste after zipped to my test project include

[image]

and it also failed to build

there is no such a directory error

what I want it just making some test program via mfc and using vs code 2013 or older version

cuz I finished test program in pythong but our CEO want it to connect opc ua in older machine based vs2013 mfc project and 2008 mfc project

and I found youtube on some guys do with linux and his comment about window

is using prebuilt pkg and some dll easy to work with that

but the prebuilt pkg don't have a dll ...

plz guide me to at least do some basic server and client to build

bbakcoin, 2023-10-12:
hello so I found old version which simply need single realesed file

https://github.com/open62541/open62541/releases?page=5
[image]

**1. my solution directory now in vs code**

[image]

**2. my c++ include**

[image]

C:\Users\amorf\Documents\Visual Studio 2015\Projects\Project1\Project1\include;%(AdditionalIncludeDirectories)

**3. my c++ code **
and  https://www.open62541.org/doc/0.3/tutorial_server_firststeps.html
I got this code form simple server example

#include <signal.h>
#include "open62541.h"

UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}

**4. the error**

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol __imp__accept@12 referenced in function _ServerNetworkLayerTCP_listen	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__bind@12 referenced in function _addServerSocket	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__closesocket@4 referenced in function _ClientNetworkLayerTCP_close	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__connect@12 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__ioctlsocket@12 referenced in function _socket_set_blocking	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__listen@8 referenced in function _addServerSocket	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__recv@16 referenced in function _connection_recv	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__select@20 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__send@16 referenced in function _connection_write	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__setsockopt@20 referenced in function _ServerNetworkLayerTCP_add	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__shutdown@8 referenced in function _ClientNetworkLayerTCP_close	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__socket@12 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__gethostname@8 referenced in function _ServerNetworkLayerTCP_start	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__WSAStartup@8 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__WSACleanup@0 referenced in function _ServerNetworkLayerTCP_stop	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__WSAGetLastError@0 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__getaddrinfo@16 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__freeaddrinfo@4 referenced in function _UA_ClientConnectionTCP	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK2019	unresolved external symbol __imp__getnameinfo@28 referenced in function _ServerNetworkLayerTCP_add	Project1	c:\Users\amorf\documents\visual studio 2015\Projects\Project1\Project1\open62541.obj	1	
Error	LNK1120	20 unresolved externals	Project1	c:\users\amorf\documents\visual studio 2015\Projects\Project1\Debug\Project1.exe	1	

hard to make what I want but I will eventually running this one in c++

but if you guys any tips and guide will be very helpful

thank you for reading

o6-nbeier, 2026-10-20:
Thank you for the question.

Use the CMake-generated library target (or add all linker dependencies yourself, including `Ws2_32.lib` for the amalgamation) and set `BUILD_SHARED_LIBS` if you need a DLL. The download is only a source archive.

Closing for now.

closed 2026-10-20 by o6-nbeier