open62541/open62541 #6061: how to deal with UA_Client_Service_write return BadNodeIdUnknown

wc220322, 2023-10-16:
hey guys!I'm worry about the problem.I use kepserver to create node and uaexpert as client,I could change nodes' value with uaexper,but fail to change the value by open62541.
----------------------------------------------------------------------------------------------------------------------------------------
code as follow:
m_nodeDatas = outputTagDataList;
    
    int arraySize = m_nodeDatas.size();

    UA_WriteValue wValueArray[arraySize];
    int i = 0;
    vector<TagInfo> tagInfoVector{ m_nodeInfos.begin(),m_nodeInfos.end() };

    //printf("m_nodeDatas size:%d\n",m_nodeDatas.size());

    for (auto it = m_nodeDatas.begin(); it != m_nodeDatas.end(); ++it)
    {       
        UA_WriteValue_init(&wValueArray[i]);
        wValueArray[i].attributeId = UA_ATTRIBUTEID_VALUE;

        string nodeValue = it->GetValue();
           string dataType = tagInfoVector[i].GetDataType();
       
            if (it->GetTagInfoId() == tagInfoVector[i].GetId())
            {            
                string dcsPosition = tagInfoVector[i].GetDCSPosition();
                char* dcsCstr = new char[dcsPosition.length() + 1];
                std::strcpy(dcsCstr, dcsPosition.c_str());
                UA_Variant variant = convertStringToVariant(nodeValue, dataType);

                wValueArray[i].nodeId = UA_NODEID_STRING((UA_UInt16)m_nameSpaceIndex, dcsCstr);
                wValueArray[i].attributeId = UA_ATTRIBUTEID_VALUE;
                wValueArray[i].value.value = variant;
                wValueArray[i].value.hasValue = true;

                printNodeId(&wValueArray[i].nodeId);

                delete[] dcsCstr;
                m_outputTagDataList.push_back(*it);
        i++;
        }
        
    }

    UA_WriteRequest writeRequest;
    UA_WriteRequest_init(&writeRequest);
    writeRequest.nodesToWrite = &wValueArray[0];
    writeRequest.nodesToWriteSize = arraySize;

    //printf("---------------- write nodes'value start-------------------\n");

    UA_WriteResponse wResp = UA_Client_Service_write(m_pClient, writeRequest);
    UA_StatusCode retval = wResp.responseHeader.serviceResult;
    for(int i = 0;i<arraySize;i++)
    {
        printUAVariant(&writeRequest.nodesToWrite[i].value.value);
    }

    if (retval == UA_STATUSCODE_GOOD)
    {

        char status;
        int i = 0;
        if (wResp.resultsSize == m_outputTagDataList.size())
        {  
            for (auto it = m_outputTagDataList.begin(); it != m_outputTagDataList.end(); ++it)
            {
                retval = wResp.results[i];
                if(retval == UA_STATUSCODE_GOOD)
                {
                    status = '1';
--------------------------------------------------------------------------------------------------------
and my retval  is BadNodeIdUnknown

jpfr, 2023-10-17:
Well, what is the output of printNodeId and ist it correct?

Note the importance of the namespace and the difference between string and bytestring.

also, different nodes might be visible depending on the login user.

wc220322, 2023-10-19:
I tried to rewrite the code and recreate the OPC NODE. Now it can be reset value normally.
thanks!>-<

closed 2023-10-19 by wc220322