open62541/open62541 #6066: What is "error/eventloop    Cannot run EventLoop from the run method itself" about?

yaalexse, 2023-10-17:
## Description
I keep getting the following error **error/eventloop    Cannot run EventLoop from the run method itself ** from the client logs using this  client code:
[examples/client_subscription_loop.c](https://github.com/open62541/open62541/blob/e71c8a4973820bdec58ff25ea6cef782e4249a4a/examples/client_subscription_loop.c)
with this server code from examples: 
[examples/tutorial_server_monitoreditems.c](https://github.com/open62541/open62541/blob/e71c8a4973820bdec58ff25ea6cef782e4249a4a/examples/tutorial_server_monitoreditems.c)
and UA_ENABLE_SUBSCRIPTION is ON.
it seems like this line:
https://github.com/open62541/open62541/blob/e71c8a4973820bdec58ff25ea6cef782e4249a4a/examples/client_subscription_loop.c#L79-L80
always returns a response with bad status code.

Logs from the client:
```
[2023-10-17 09:52:50.626 (UTC+0200)] info/network       TCP 4   | New connection to "localhost" on port 4840
[2023-10-17 09:52:50.626 (UTC+0200)] info/network       TCP 5   | Socket closed
[2023-10-17 09:52:50.626 (UTC+0200)] info/userland      Waiting for ack
[2023-10-17 09:52:50.626 (UTC+0200)] info/userland      Waiting for OPN Response
[2023-10-17 09:52:50.626 (UTC+0200)] info/channel       TCP 4   | SC 3714       | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-17 09:52:50.626 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-10-17 09:52:50.626 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2023-10-17 09:52:50.626 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2023-10-17 09:52:50.626 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-10-17 09:52:50.626 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2023-10-17 09:52:50.626 (UTC+0200)] info/userland      A session with the server is activated
[2023-10-17 09:52:50.626 (UTC+0200)] error/eventloop    Cannot run EventLoop from the run method itself
[2023-10-17 09:52:50.627 (UTC+0200)] warn/client        Request with unknown RequestId 2982
[2023-10-17 09:52:50.627 (UTC+0200)] error/client       Processing the message returned the error code BadSecurityChecksFailed
[2023-10-17 09:52:50.627 (UTC+0200)] warn/client        Client connection already initiated
[2023-10-17 09:52:50.627 (UTC+0200)] info/client        Client Status: ChannelState: Closing, SessionState: Created, ConnectStatus: Good
[2023-10-17 09:52:50.627 (UTC+0200)] info/channel       TCP 0   | SC 0  | Closed the SecureChannel
[2023-10-17 09:52:50.627 (UTC+0200)] warn/client        skip verifying ApplicationURI for the SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None
[2023-10-17 09:52:50.627 (UTC+0200)] info/network       TCP 5   | New connection to "localhost" on port 4840
[2023-10-17 09:52:50.627 (UTC+0200)] info/network       TCP 4   | Socket closed
[2023-10-17 09:52:50.627 (UTC+0200)] info/userland      Waiting for ack
[2023-10-17 09:52:50.627 (UTC+0200)] info/userland      Waiting for OPN Response
[2023-10-17 09:52:50.627 (UTC+0200)] info/channel       TCP 5   | SC 3715       | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-17 09:52:50.627 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Created, ConnectStatus: Good
[2023-10-17 09:52:50.627 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2023-10-17 09:52:50.627 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2023-10-17 09:52:50.627 (UTC+0200)] info/client        Client Status: ChannelState: Open, SessionState: Activated, ConnectStatus: Good
[2023-10-17 09:52:50.627 (UTC+0200)] info/userland      A SecureChannel to the server is open
[2023-10-17 09:52:50.627 (UTC+0200)] info/userland      A session with the server is activated
[2023-10-17 09:52:50.627 (UTC+0200)] error/eventloop    Cannot run EventLoop from the run method itself
[2023-10-17 09:52:50.629 (UTC+0200)] warn/client        Request with unknown RequestId 2986
[2023-10-17 09:52:50.629 (UTC+0200)] error/client       Processing the message returned the error code BadSecurityChecksFailed
[2023-10-17 09:52:50.629 (UTC+0200)] warn/client        Client connection already initiated
[2023-10-17 09:52:50.629 (UTC+0200)] info/client        Client Status: ChannelState: Closing, SessionState: Created, ConnectStatus: Good
```

Logs on server side:
```
[2023-10-17 10:19:35.990 (UTC+0200)] info/channel       TCP 9   | SC 410        | SecureChannel created
[2023-10-17 10:19:35.990 (UTC+0200)] info/channel       TCP 0   | SC 0  | SecureChannel closed
[2023-10-17 10:19:35.990 (UTC+0200)] info/network       TCP 10  | Socket closed
[2023-10-17 10:19:35.990 (UTC+0200)] info/channel       TCP 9   | SC 410        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-17 10:19:35.990 (UTC+0200)] warn/client        Could not find a SecurityPolicy with encryption for the UserTokenPolicy. Using an unencrypted policy.
[2023-10-17 10:19:35.990 (UTC+0200)] info/session       TCP 9   | SC 410        | Session "ns=1;g=374f8057-9288-37de-12a1-5af5614614c1"   | ActivateSession: Session attached to new channel
[2023-10-17 10:19:35.990 (UTC+0200)] info/session       TCP 9   | SC 410        | Session "ns=1;g=374f8057-9288-37de-12a1-5af5614614c1"   | ActivateSession: Session activated with ClientUserId ""
[2023-10-17 10:19:35.993 (UTC+0200)] info/session       TCP 9   | SC 410        | Session "ns=1;g=374f8057-9288-37de-12a1-5af5614614c1"   | Subscription 410 | Subscription created (Publishing interval 500.00ms, max 1000 notifications per publish)
[2023-10-17 10:19:35.994 (UTC+0200)] info/network       TCP 10  | Connection opened from "127.0.0.1" via the server socket 5
[2023-10-17 10:19:35.994 (UTC+0200)] info/channel       TCP 10  | SC 411        | SecureChannel created
[2023-10-17 10:19:35.994 (UTC+0200)] info/channel       TCP 0   | SC 0  | SecureChannel closed
[2023-10-17 10:19:35.994 (UTC+0200)] info/network       TCP 9   | Socket closed
[2023-10-17 10:19:35.994 (UTC+0200)] info/channel       TCP 10  | SC 411        | SecureChannel opened with SecurityPolicy http://opcfoundation.org/UA/SecurityPolicy#None and a revised lifetime of 600.00s
[2023-10-17 10:19:35.994 (UTC+0200)] warn/client        Could not find a SecurityPolicy with encryption for the UserTokenPolicy. Using an unencrypted policy.
[2023-10-17 10:19:35.994 (UTC+0200)] info/session       TCP 10  | SC 411        | Session "ns=1;g=374f8057-9288-37de-12a1-5af5614614c1"   | ActivateSession: Session attached to new channel
[2023-10-17 10:19:35.994 (UTC+0200)] info/session       TCP 10  | SC 411        | Session "ns=1;g=374f8057-9288-37de-12a1-5af5614614c1"   | ActivateSession: Session activated with ClientUserId ""
[2023-10-17 10:19:35.996 (UTC+0200)] info/session       TCP 10  | SC 411        | Session "ns=1;g=374f8057-9288-37de-12a1-5af5614614c1"   | Subscription 411 | Subscription created (Publishing interval 500.00ms, max 1000 notifications per publish)
```

Used CMake options:

```bash
cmake ..
```

## Checklist

 - [x] open62541 Version (release number or git tag): 1.3.8-1356-g744ea7edf
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 22.04
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-10-17:
The event loop drives the control flow of the server. In order to be responsive, only asynchronous operations are permitted from within the event loop.

so you can either use the async version of the api. Or you return to the top caller that drops into the client control flow.

yaalexse, 2023-10-26:
Thank you for the response, by using the async method it worked fine.

closed 2023-10-26 by yaalexse

Soroush-Shariaty, 2024-05-12:
Hello. I have the same problem and found this link.
By " use the async version of the api" you mean using the `UA_Client_Subscriptions_create_async` method?
What other changes should I make for it to work?
Thanks in advance.