open62541/open62541 #6070: Syntax errors compiling open62541.c with eCos 3.0

Isarhamster, 2023-10-18:
## Description
Compiling open62541 with eCos 3.0 runs into syntax errors: 

```
error: 'for' loop initial declarations are only allowed in C99 or C11 mode
         for(size_t i = 0; i < customTypes->typesSize; ++i) {

```
## Background Information / Reproduction Steps
open62541 uses (unnecessarily) extensions to the C programming language which came in with C99. This makes porting into embedded environments with eCos difficult. Especially the following pattern:

`for(size_t i=0; ...)`
instead of 
`size_t i; for(i=0;...)`
in more than 400 locations makes  open62541 hard to use with gcc.

eCos3.0 applications must be compiled with `gcc -std=c89` if they use the standard library thus using `gcc -std=c99` (which would mask the syntax errors in open62541) is not an option.

Used CMake options:

cmake -DUA_NAMESPACE_ZERO=FULL -DUA_ARCHITECTURE=eCos -DUA_ENABLE_AMALGAMATION=ON ../

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): master (1.3.8)
 - [X] Other OPC UA SDKs used (client or server): None
 - [X] Operating system: eCos
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-10-19:
This project uses C99 since the beginning about 10 years ago.
To switch to C89 we would have to rewrite every method, every loop and nearly every variable declaration.
Also this is the first time in a long time that we get this request.

WONTFIX.

You compiler can do both C89 and C99.
This is not a problem of the compiler, but a problem of telling it what to allow.
You can add additional `-WnoXXX` gcc flags for the warnings you encounter.

Or you put the interaction with the standard library in a shallow wrapper.
And use the C89 mode to create an object file for the wrapper.
This can then be linked with everything else compiler as C99.
Most non-trivial interactions with the stdlib are already behind some `UA_xxx` macro in the ua_architecture files.

Isarhamster, 2023-10-20:
I agree that all for loops are a horrible task - but would it be possible to change the two occurrences inside the open62541 header? This would allow compiling open62541.c with -C99 while compiling the remaining code with -C89. 
There are more items I need to modify inside open62541.h, but this would make porting/maintanance more transparent for future developers.

jpfr, 2023-10-20:
Sure, if we can make the headers C89 ready with little effort, then we can do so.
You can paste a patch here or open a Pull Request.

Isarhamster, 2023-11-07:
I opened a Pull Request for this (together with 3 other items I modified in my code - I wanted to open separate Pull Requests for them as they are not really related - but my knowledge in Git was not sufficient to accomplish this)