open62541/open62541 #6071: Stuck in UA_EventLoopPOSIX_pollFDs

JakobKlocker, 2023-10-19:
When trying to connect to a invalid address, there's an endless loop. I tried debugging, timeout is set to 5000.

Code is following:
`   UA_Client *client = UA_Client_new();
    UA_ClientConfig_setDefault(UA_Client_getConfig(client));
    UA_StatusCode retval = UA_Client_connect(client, this->opcuaUrl.c_str());
`
When I force quit (ctrl + c) this error message shows:
`
[2023-10-19 11:22:18.366 (UTC+0200)] warn/eventloop     Timeout during poll
`
These messages were shown before the one above
`
[2023-10-19 11:25:06.699 (UTC+0200)] info/client        Client Status: ChannelState: Fresh, SessionState: Closed, ConnectStatus: BadConnectionRejected
[2023-10-19 11:25:06.700 (UTC+0200)] info/network       TCP 5   | Socket closed
`

When debuging, it seems to be stuck in function UA_EventLoopPOSIX_pollFDs at following line
`
int events = epoll_wait(epollfd, epoll_events, 64,
                            (int)(listenTimeout / UA_DATETIME_MSEC));
`

What's weird as well, if I step over manually over the line in the debugger it conitnues. I'm not sure how that's possible.

 - [X ] open62541 Version (release number or git tag): v1.3.8
 - [X] Operating system: Ubuntu/WSL

jpfr, 2023-10-26:
I suspect this has something to do with DNS lookup.
We might have to change to an async DNS lookup outside of the typical POSIX API.

Are you on the 1.3 branch (as indicated) or on 1.4/master?
I'm asking because the epoll call looks rather like something we would do on a more recent development version.

JakobKlocker, 2023-11-03:
I've actually compiled the master, after using the version 1.3.8 it seems to work.
Thanks a lot for pointing that out!

closed 2023-11-03 by JakobKlocker