open62541/open62541 #6083: Incorrect computation of Thumbprint in asymmetric algorithm security header

phuong-nm, 2023-10-23:
## Description

ReceiverCertificateThumbprint shall be computed over sender application instance certificate. This is the leaf certificate if the sender sends a chain of certificates.
The open62541 stack, however, computes the thumbprint over the complete chain of certificates.

[image]

## Background Information / Reproduction Steps

- A open62541-based Server is provided with a certificate that is signed by a root CA.
- A OPC UA client is provided with a certificate that is signed by the same root CA.
- The client attempts to connect to the server. In SenderCertificate field of CreateSecureChannelRequest the client provides the complete chain of certificate.
- The server replies with a CreateSecureChannelResponse without error. However, the ReceiverCertificateThumbprint field of the CreateSecureChannelResponse is computed over the complete chain of certificate. This is incorrect according to "OPC 10000-6: Mappings, Release V1.04, Table 42 Asymmetric algorithm Security header".

## Suggested solution

Only Application Instance Certificate of the sender should be used in setting security policy UA_SecureChannel_setSecurityPolicy. As the open62541 does not handle chain of certificates in handling the remote certificate after message verification, it should simply ignore extra bytes in the chain except for the leaf certificate.

```c
UA_StatusCode
UA_Server_configSecureChannel(void *application, UA_SecureChannel *channel,
                              const UA_AsymmetricAlgorithmSecurityHeader *asymHeader) {
    /* Iterate over available endpoints and choose the correct one */
    UA_SecurityPolicy *securityPolicy = NULL;
    UA_Server *const server = (UA_Server *const) application;
    for(size_t i = 0; i < server->config.securityPoliciesSize; ++i) {
        UA_SecurityPolicy *policy = &server->config.securityPolicies[i];
        if(!UA_ByteString_equal(&asymHeader->securityPolicyUri, &policy->policyUri))
            continue;

        UA_StatusCode retval = policy->asymmetricModule.
            compareCertificateThumbprint(policy, &asymHeader->receiverCertificateThumbprint);
        if(retval != UA_STATUSCODE_GOOD)
            continue;

        /* We found the correct policy (except for security mode). The endpoint
         * needs to be selected by the client / server to match the security
         * mode in the endpoint for the session. */
        securityPolicy = policy;
        break;
    }

    if(!securityPolicy)
        return UA_STATUSCODE_BADSECURITYPOLICYREJECTED;

    /*** Hilscher ***/
    /* If the sender provides a chain of certificates then we shall extract the ApplicationInstanceCertificate.
    * and ignore the extra bytes.
    * See also: OPC UA Part 6, V1.04, 6.7.2.3 Security Header, Table 42 - Asymmetric algorithm Security header */
    UA_ByteString senderApplicationInstanceCertificate = UA_BYTESTRING_NULL;
    getLeafCertificate(&asymHeader->senderCertificate, &senderApplicationInstanceCertificate);
    /*** Hilscher ***/

    /* Create the channel context and parse the sender (remote) certificate used for the
     * secureChannel. */
    UA_StatusCode retval =
        UA_SecureChannel_setSecurityPolicy(channel, securityPolicy,
                                           &senderApplicationInstanceCertificate); /*** Hilscher ***/
    if(retval != UA_STATUSCODE_GOOD)
        return retval;

    channel->securityToken.tokenId = server->lastTokenId++;
    return UA_STATUSCODE_GOOD;
}
``` 

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): V1.3.6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

jpfr, 2023-10-26:
Good find, thanks.
Do we have permission to use the code you pasted above?
If you want, you can also name a copyright notice to include.

mlang-de, 2023-10-27:
Hello @jpfr ,
my collegue @phuong-nm is on vacation for the next weeks,
therefore the approval from my side.

"Do we have permission to use the code you pasted above?"
Yes, feel free to copy the pasted code.

"If you want, you can also name a copyright notice to include."
I'm pretty sure he would be happy to be in the copyright notes :).

Copyright 2023 (c) Hilscher Gesellschaft für Systemautomation mbH (Author: Phuong Nguyen)

BR

jpfr, 2023-11-28:
👍 
This now goes to 1.3 via #6146.

closed 2023-11-29 by jpfr