open62541/open62541 #6091: InputArguments has only CurrentRead

m-obi, 2023-10-25:
## Description
Shouldn't have InputArguments also the CurrentWrite flag set in AccessLevel? With CurrentRead only, the user is not allowed to set the input arguments for a method call.

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): master
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Win10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

basyskom-jvoe, 2023-10-25:
The InputArguments and OutputArguments properties only define the arguments a method expects/returns and should not be written by clients.
For a method call, the input arguments are part of the request.

m-obi, 2023-10-25:
Ahh yes ok. Understood.

closed 2023-10-25 by m-obi