open62541/open62541 #6093: handle predefined structs with std::string/CString members

merlin81, 2023-10-26:
Hi

I've some predefined structs with std::string/CString members like

```struct Foo
{
  int i; 
  std::string str;
  CString cstr;
} 
```

What does the UA_DataTypeMembers have to look like in order to encode/decode it correctly as strings ?

```
{
   UA_TYPENAME("i").           /* .memberName */
   &UA_TYPES[UA_TYPES_INT32],  /* .memberType */
   0,                          /* .padding */
   false,                      /* .isArray */
   false.                      /* .isOptional */
},
{
   UA_TYPENAME("str").         /* .memberName */
   &UA_TYPES[UA_TYPES_???],    /* .memberType */
   0,                          /* .padding */
   false/true?,                /* .isArray */
   false.                      /* .isOptional */
},
{
   UA_TYPENAME("cstr").        /* .memberName */
   &UA_TYPES[UA_TYPES_???],    /* .memberType */
   0,                          /* .padding */
   false/true?,                /* .isArray */
   false.                      /* .isOptional */
}
```
thx in advance !

jpfr, 2023-10-26:
We have the type UA_String and UA_ByteString.
Both are "just" a length field and a pointer to a byte-array.
The conversion to and from std::string should be straightforward.

merlin81, 2023-10-27:
thx for reply,

I know the data types, but that would mean I need mirror structures (only for opcua) that looks the same except for the strings and i have to convert every string from std::string/cstring to ua_string and vice versa before sending and after receiving from the server.

As a part of a proof of concept, there are several hundred structures, some 20 levels or more deep, with arrays, etc. So translating every single structure is not an option.

I'm looking for a way to describe (`UA_DataType`/`UA_DataTypeMember`)  these predefined structures and translate it in one go with `UA_Variant_setScalar` for sending to server and cast the variant back to structure after reading from server. 

Is it possible to define a `UA_CString` or `UA_StdString` (without the length-member like UA_String-structure) ? Important is also the right encoding/decoding. A bytestring wouldn't be interpreted as string (e.g. in uaexpert), right ?

maybe with ?
- [https://www.open62541.org/doc/1.3/types.html#generic-type-handling](url) 
- [https://www.open62541.org/doc/1.3/types.html#encoding-decoding](url) 

Are there some examples to define another "buildin type" like CString/std::string ?
[examples/custom_datatype](url) didn't help in my case.

thx in advance !

jpfr, 2023-10-30:
Hey there,

we need structures expressed in terms of the OPC UA type system.
Otherwise they cannot be encoded in network messages.

That is you need to translate the structures that you have into
- 25 base types (Strings, Integers, Variants, etc.)
- Structure built up from the the base types
- Arrays
- Unions

Part 6 of the spec explains the type systems.
And you could have a look at the code for `UA_copy` that basically covers all of the above under the hood.
https://github.com/open62541/open62541/blob/effaeb395bded7e31f62ad4d7001c9ded8f2ae60/src/ua_types.c#L1415

I think it would be easiest if you define conversions for the "base types" and then automate/code-generate the conversion of structures on top.

merlin81, 2023-10-31:
it doesn't look like an easy task to add another universal and stable layer for automated conversion between nested null terminated-strings and ua_strings, but I will try your suggestion ... thx for reply 

i'm thinking about a recursion through the `UA_DataType` with some (dummy) datatypes (std::string, cstring) to trigger the conversion.  so every structure will have 2 datatypes. one for cstring-use, and a derived one for opcua-use. If anyone has a better idea, please let me know

----------------------

... I don't often have non-null terminated string-datatypes in C++ (C++ is common use in industry). so I think several people should have the same problem with this gap between existing code and opcua type system. rewriting existing data structures (and dependent code/subsystems/...) in established and tested code could be a co-criterion for switching to opcua. opcua/open62541 could be very powerful tools to solve today's requirements in professional use. was there no consideration for null-terminated strings when opc-foundation developing Part 6?