open62541/open62541 #6097: Extreme memory footprint comparing 1.3.8 with 1.2

Isarhamster, 2023-10-27:
## Description
open62541 in version 1.3.8 has an extreme memory footprint, especially on the stack. While version 1.2 was running with setting the stack size to 0x5000, version 1.3.8 requires a stack of at least 0x30000 - that is almost 10x the size needed in 1.2.  

## Background Information / Reproduction Steps
I get different issues depending on the namespace settings:
In 1.3.8, only setting -DUA_NAMESPACE_ZERO=FULL creates a UA-stack that can execute the function `namespace0_generated()` without crashing. All other UA stacks (compiled with REDUCED or NULL) will crash in `namespace0_generated()`.
My embedded system has only 12MB RAM available for heap and stack. This was no issue in 1.2 though.
Inspecting the compilation reveals that
- when compiling with FULL, there is a method 
`function_namespace0_generated_3351_begin` which uses 133128 bytes static stack - and this is just one of the methods called in namespace0_generated. 
- when compiling with REDUCED, there is a variable `static const UA_Byte variablenode_ns_0_i_7617_variant_DataContents_byteArray[**177218**]`  and I have a feeling, that this variable gets copied into either stack and/or heap. This is not the only issue, but I have a hard time debugging the stack as I simple get a vector catch (which is slightly too late to debug this).

Is there some define to get back the gentle behaviour of version 1.2?
Version 1.2 had `function_namespace0_generated_164_begin()` as largest stack user with just 1936 bytes of stack needed.

Used CMake options:
On 1.3.8 either
>cmake -DUA_NAMESPACE_ZERO=FULL -DUA_ARCHITECTURE=eCos -DUA_ENABLE_AMALGAMATION=ON ../
or
>cmake -DUA_NAMESPACE_ZERO=REDUCED -DUA_ARCHITECTURE=eCos -DUA_ENABLE_AMALGAMATION=ON ../

On 1.2 and 1.3.8:
>cmake -DUA_ARCHITECTURE=eCos -DUA_ENABLE_AMALGAMATION=ON ../

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: eCos
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue (at least for me)

r-sin, 2023-11-14:
Have a look at #5893, might be related.

Isarhamster, 2023-11-21:
I missed that issue as my keyword for the search was "stack". Thanks for pointing out the related issue.