open62541/open62541 #6101: Adding Machinery information model fails with error 'Reference type to the parent is not hierarchical'

HHVB-17, 2023-10-30:
## Description
Adding of Machinery Information Model to the Open62541 OPC UA Server fails with the error 'Reference type to the parent is not hierarchical'.

I have used the CMake file similar to the documentation to generate the datatype of DI and nodeset for DI, Machinery, IRDI and PADIM. The datatype and nodeset files for all the above mentioned information models are successfully generated without any error. But when I include these files in the OPC UA Server source code and call these namespaces in the main program, the error arises for only the namespace Machinery and PADIM while executing the program.

## Background Information / Reproduction Steps
The CMake file that is used to only create the data types and nodesets from the information model is provided below

```bash
cmake_minimum_required(VERSION 3.0...3.12)
# Project name definition
project(GB_Nodeset_Generator) 

if(${CMAKE_VERSION} VERSION_LESS 3.12)
    cmake_policy(VERSION ${CMAKE_MAJOR_VERSION}.${CMAKE_MINOR_VERSION})
endif()

# Project configuration
if(CMAKE_PROJECT_NAME STREQUAL PROJECT_NAME)
    if(UA_NAMESPACE_ZERO STREQUAL "FULL")
        find_package(open62541 REQUIRED COMPONENTS FullNamespace)
    else()
        find_package(open62541 REQUIRED)
    endif()

    function(assign_source_group)
        # define empty function. We don't need it in standalone
    endfunction(assign_source_group)

    include_directories(${PROJECT_BINARY_DIR}/src_generated)
endif()

if(NOT CMAKE_PROJECT_NAME STREQUAL PROJECT_NAME)
    # needed or cmake doesn't recognize dependencies of generated files
    set(PROJECT_BINARY_DIR ${CMAKE_BINARY_DIR})
endif()

if(UA_ENABLE_AMALGAMATION)
    add_definitions(-DUA_ENABLE_AMALGAMATION)
endif()

# Assign UA_NS_DIR
set(UA_NS_DIR ${PROJECT_SOURCE_DIR}/deps/ua-nodeset)

# Data type generation for namespace "DI"
ua_generate_datatypes(
  NAME "types_di"
  TARGET_SUFFIX "types-di"
  FILE_NS "${UA_NS_DIR}/DI/Opc.Ua.Di.NodeSet2.xml"
  FILE_CSV "${UA_NS_DIR}/DI/Opc.Ua.Di.NodeIds.csv"
  FILES_BSD "${UA_NS_DIR}/DI/Opc.Ua.Di.Types.bsd"
  NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
  OUTPUT_DIR "${OUTPUT_DIR}"
)
# Nodeset generation for namespace "DI"
ua_generate_nodeset(
    NAME "di"
    FILE "${UA_NS_DIR}/DI/Opc.Ua.Di.NodeSet2.xml"
    TYPES_ARRAY "UA_TYPES_DI"
    INTERNAL
    DEPENDS_TYPES "UA_TYPES"
    DEPENDS_NS "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml"
    DEPENDS_TARGET "open62541-generator-types-di"
)

# Nodeset generation for namespace "machinery"
ua_generate_nodeset(
    NAME "machinery"
    FILE "${UA_NS_DIR}/Machinery/Opc.Ua.Machinery.NodeSet2.xml"
    INTERNAL
    DEPENDS_TYPES
        "UA_TYPES" 
    DEPENDS_NS
        "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml"
        "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml"
    DEPENDS_TARGET "open62541-generator-ns-di"
)

# Nodeset generation for namespace "IRDI"
ua_generate_nodeset(
    NAME "IRDI"
    FILE "${PROJECT_SOURCE_DIR}/nodeset/IRDI.xml"
    DEPENDS_TYPES 
        "UA_TYPES" 
    INTERNAL
    DEPENDS_NS
        "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml"
)        

# Nodeset generation for namespace "PADIM"
ua_generate_nodeset(
    NAME "PADIM"
    FILE "${PROJECT_SOURCE_DIR}/nodeset/PADIM.xml"
    DEPENDS_TYPES 
        "UA_TYPES"
    INTERNAL
    DEPENDS_NS 
        "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml"
        "${PROJECT_SOURCE_DIR}/deps/ua-nodeset/DI/Opc.Ua.Di.NodeSet2.xml"
        "${PROJECT_SOURCE_DIR}/nodeset/IRDI.xml"
    DEPENDS_TARGET "open62541-generator-ns-IRDI"  "open62541-generator-ns-machinery"
)

# Create a custom target for GB namespace generation
add_custom_target(generate_GB_nodeset
  DEPENDS open62541-generator-ns-PADIM
)

```
After successful generation of data type and nodeset files, these files were included in the OPC UA Server source code as shown below.

```bash
int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server* server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));
 
    UA_StatusCode retval = di(server);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
            "Adding the DI namespace failed. Please check previous error output.");
        UA_Server_delete(server);
        return (int)UA_STATUSCODE_BADUNEXPECTEDERROR;
    }
 
    retval = machinery(server);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
            "Adding the machinery namespace failed. Please check previous error output.");
        UA_Server_delete(server);
        return (int)UA_STATUSCODE_BADUNEXPECTEDERROR;
    }

    retval = irdi(server);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
            "Adding the irdi namespace failed. Please check previous error output.");
        UA_Server_delete(server);
        return (int)UA_STATUSCODE_BADUNEXPECTEDERROR;
    }

    retval = padim(server);
    if (retval != UA_STATUSCODE_GOOD) {
        UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_SERVER,
            "Adding the padim namespace failed. Please check previous error output.");
        UA_Server_delete(server);
        return (int)UA_STATUSCODE_BADUNEXPECTEDERROR;
    }

    retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```
And when executing the program, the following error occurs as shown in the below photo

[image]

In the above case the DI namespace is successfully added to the OPC UA Server. But the error occurs when the Machinery namespace is to be added to the server. So I tried adding the IRDI namespace by skipping the Machinery namespace to the Server and it was successfully added to the server along with DI namespace. But adding the following PADIM namespace to the server failed as shown in the below image

[image]

Since both the Machinery and PADIM are dependent on DI namespace and throws up the similar error, I created the nodeset for AutoID Information model which also depends on DI. Then I added the AutoID namespace to the OPC UA Server along with DI and this works successfully without any error.

Therefore, kindly help me to figure out the reason for the errors while adding the Machinery and PADIM namespace to the OPC UA Server.

Used CMake options:

```bash
-DCMAKE_CONFIGURATION_TYPES:STRING="Debug;Release;MinSizeRel;RelWithDebInfo" -DUA_NAMESPACE_ZERO:STRING="FULL" 
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): V1.3.7
 - [x] Other OPC UA SDKs used (client or server):  UA Expert Client
 - [x] Operating system: Windows 11
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [x] Critical issue

ccvca, 2023-10-31:
I do suspect, that the Machinery requiers a newer version of the Ns0. As you are using 1.3.7, the linkes Nodesets for NS0 are from the october 19, 2021[deps/ua-nodeset](https://github.com/OPCFoundation/UA-Nodeset/tree/f71b3f411d5cb16097c3ae0c744f67ad45535ffb).

At least the recent Version of open62541 works fine with all namespaces in [this project](https://github.com/umati/Sample-Server):
https://github.com/umati/Sample-Server/blob/66fdc2437c82cc21ab8789519526ce54a1e832a9/CMakeLists.txt

https://github.com/umati/Sample-Server/blob/66fdc2437c82cc21ab8789519526ce54a1e832a9/SampleServer.cpp#L224-L236

Schievel1, 2025-02-10:
This is correct. The `HasAddin` reference changed, and now one of it's references is hierarchical, which the node 5002 machinery set requires. This usually happens when you use the reduced nodeset from open62541 instead of the full. The reduced nodeset is a bit behind full.