open62541/open62541 #6102: Streaming video on OPC UA

sdfzz, 2023-10-31:
## Description
I want to stream video over OPC UA.

## Background Information / Reproduction Steps
Hi all, 

I'm trying to stream video over OPC UA. 

Basically, my OPCUA server has an image variable, which is updated periodically with newest image from a camera. 

So, if a client subscribes it, it can see the video stream. 

I'm using Qt to build an OPCUA server, and using UaExpert as the client.

1. What kind of data type should an image variable have? I tried UA_TYPES_STRING, and I got 'unsupported node dropped (ValueRank = -2, DataType=BaseDataType)' error in UaExpert's ImageViewer plugin
2. Is there any example or tutorial that I can follow? 

Any help would be appreciated

Regards,

Steve

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

sdfzz, 2023-10-31:
@undroidwish Thank you for the link! I will look into it, although I'm not familiar with TCL...

I guess I have to spend more time on this.

For one thing, How does UaExpert knows if a node is image? Does it know it by data type (bytestring)? or from name space (Image)?

I'm also attaching my code so that in case someone can help me out on this...

This is how I'm adding bytestring variable to server:
```
  UA_VariableAttributes attr = UA_VariableAttributes_default;
  UA_Variant_setScalar(&attr.value, &attrVal, &UA_TYPES[UA_TYPES_BYTESTRING]);
  attr.accessLevel = (UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE);
  UA_Server_addVariableNode(server, UA_NODEID_STRING(2, "Image"), *parent,
                UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                UA_QUALIFIEDNAME(2, "Camera Image"),
                UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
```
And I'm updating image data by:
```
UA_ByteString img = UA_BYTESTRING(rgb_img_data.data());
UA_Variant_setArray(&value, &img, 1, &UA_TYPES[UA_TYPES_BYTESTRING]);
UA_Server_writeValue(g_opcua_server, UA_NODEID_STRING(2, "Image"), value);
```
Note that I'm using Qt, so my image (rgb_img_data) is in QByteArray format.

Regards,

Steve

undroidwish, 2023-10-31:
Use an ImageBMP, ImagePNG, ImageGIF, or ImageJPG datatype which is a subtype of ByteString.
Write BMP, PNG, GIF, or JPG data into it. And UAExport should be able to do the Right Thing(tm).

sdfzz, 2023-10-31:
Sorry, but where is ImagePNG defined?

I get:

error: use of undeclared identifier 'UA_TYPES_IMAGEPNG'

Sorry for frequent questions

Steve

undroidwish, 2023-10-31:
Just write a UA_TYPES_BYTESTRING since the Image* things are subtyped from ByteString.

sdfzz, 2023-11-01:
@undroidwish Thank for reply

The mistake was that my dataType was not ByteString

Once I fixed it, I can drag-and-drop node into image viewer of UaExpert

No image is showing up, but I think it is more related to image format 

Regards,

Steve

closed 2026-10-20 by o6-nbeier