open62541/open62541 #6103: Cannot find UA_Server_runUntilInterrupt when building with single file distribution

KUR-creative, 2023-11-01:
## Description
I tried to build example [server code](https://github.com/open62541/open62541#examples) in README with single file distribution [v1.3.6](https://github.com/open62541/open62541/releases/tag/v1.3.6). However, build failed due to an undefined reference.

myserver.c
```c
#include "open62541.h"

int main(int argc, char** argv)
{
    /* Create a server listening on port 4840 (default) */
    UA_Server *server = UA_Server_new();

    /* Add a variable node to the server */

    /* 1) Define the variable attributes */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    attr.displayName = UA_LOCALIZEDTEXT("en-US", "the answer");
    UA_Int32 myInteger = 42;
    UA_Variant_setScalar(&attr.value, &myInteger, &UA_TYPES[UA_TYPES_INT32]);

    /* 2) Define where the node shall be added with which browsename */
    UA_NodeId newNodeId = UA_NODEID_STRING(1, "the.answer");
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_NodeId variableType = UA_NODEID_NULL; /* take the default variable type */
    UA_QualifiedName browseName = UA_QUALIFIEDNAME(1, "the answer");

    /* 3) Add the node */
    UA_Server_addVariableNode(server, newNodeId, parentNodeId,
                              parentReferenceNodeId, browseName,
                              variableType, attr, NULL, NULL);

    /* Run the server (until ctrl-c interrupt) */
    UA_StatusCode status = UA_Server_runUntilInterrupt(server);

    /* Clean up */
    UA_Server_delete(server);
    return status == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

```
PS C:\Users\kur\Desktop\proj\opc-ua\on_same_host> gcc -std=c99 -lws2_32 -lwsock32 -DUA_ARCHITECTURE_WIN32 .\open62541.c \open62541.h .\myserver.c -o server
.\myserver.c: In function 'main':
.\myserver.c:29:28: warning: implicit declaration of function 'UA_Server_runUntilInterrupt' [-Wimplicit-function-declaration]
   29 |     UA_StatusCode status = UA_Server_runUntilInterrupt(server);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbb833): undefined reference to `__imp_getaddrinfo'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbb868): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbb94f): undefined reference to `__imp_inet_pton'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbb966): undefined reference to `__imp_ntohl'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbb9d7): undefined reference to `__imp_inet_pton'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbba49): undefined reference to `__imp_socket'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbabb): undefined reference to `__imp_freeaddrinfo'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbba9): undefined reference to `__imp_setsockopt'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbbc2): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbc44): undefined reference to `__imp_closesocket'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbc57): undefined reference to `__imp_freeaddrinfo'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbcd1): undefined reference to `__imp_setsockopt'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbce6): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbd9f): undefined reference to `__imp_setsockopt'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbdb4): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbe63): undefined reference to `__imp_inet_pton'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbe7f): undefined reference to `__imp_htonl'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbeb2): undefined reference to `__imp_inet_pton'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbfaa): undefined reference to `__imp_inet_pton'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbbff1): undefined reference to `if_nametoindex'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbc085): undefined reference to `__imp_setsockopt'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbc09e): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbc17e): undefined reference to `__imp_bind'
[... 87 lines cut ...]
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbf709): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbfbed): undefined reference to `__imp_getaddrinfo'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.text+0xbfc21): undefined reference to `__imp_WSAGetLastError'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccKtrbz7.o:open62541.c:(.rdata$.refptr.in6addr_any[.refptr.in6addr_any]+0x0): undefined reference to `in6addr_any'
C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\kur\AppData\Local\Temp\ccDhy78S.o:myserver.c:(.text+0x789): undefined reference to `UA_Server_runUntilInterrupt'
collect2.exe: error: ld returned 1 exit status
```

After searching, I found that UA_Server_runUntilInterrupt was not defined in the open62541.h file.

Doesn't single file distribution include plugins? How should build the example using single file distribution?

thank you

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

basyskom-jvoe, 2023-11-01:
The UA_Server_runUntilInterrupt() function is not part of open62541 1.3.x, the first release with it will be 1.4.
Your link leads to the README of the master branch. This is how it's done with 1.3.x: https://github.com/open62541/open62541/tree/1.3#examples

KUR-creative, 2023-11-02:
Thanks. It works perfectly! 

```bash
gcc -std=c99 -DUA_ARCHITECTURE_WIN32 .\open62541.c .\myserver.c -o server -lws2_32 -lIphlpapi
```

closed 2023-11-02 by KUR-creative