open62541/open62541 #6133: UA_Client_connect can't be run in loop.

ila162, 2023-11-17:
## Description
UA_Client_connect in the newest version can't be run in loop, the function clear endpointUrl and discoveryUrl, and only fill the endpointUrl (this is fine in first run the function when responseFindServers fills discoveryUrl), which leads to return UA_STATUSCODE_BADTIMEOUT from connectSync because of failure to meet the requirement client->discoveryUrl.length > 0.

```c++
UA_StatusCode
UA_Client_connect(UA_Client *client, const char *endpointUrl) {
    /* Set the endpoint URL the client connects to */
    UA_String_clear(&client->endpointUrl);
    UA_String_clear(&client->discoveryUrl);
    client->endpointUrl = UA_STRING_ALLOC(endpointUrl);

    /* Open a Session when possible */
    client->noSession = false;

    /* Connect Synchronous */
    return connectSync(client);
}
```

```c++
UA_StatusCode
connectSync(UA_Client *client) {
    UA_DateTime now = UA_DateTime_nowMonotonic();
    UA_DateTime maxDate = now + ((UA_DateTime)client->config.timeout * UA_DATETIME_MSEC);

    UA_StatusCode retval = initConnect(client);
    if(retval != UA_STATUSCODE_GOOD)
        return retval;

    while(retval == UA_STATUSCODE_GOOD) {
        if(!client->endpointsHandshake && !client->findServersHandshake &&
           client->discoveryUrl.length > 0 &&
           (client->sessionState == UA_SESSIONSTATE_ACTIVATED ||
            (client->noSession && client->channel.state == UA_SECURECHANNELSTATE_OPEN)))
            break;
        now = UA_DateTime_nowMonotonic();
        if(maxDate < now)
            return UA_STATUSCODE_BADTIMEOUT;
        retval = UA_Client_run_iterate(client,
                                       (UA_UInt32)((maxDate - now) / UA_DATETIME_MSEC));
    }

    return retval;
}
```
## Background Information / Reproduction Steps
This can be reproduce in client_connect_loop.c example.

## Checklist

Please provide the following information:

 - [ 1.3.8] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-11-28:
This is a blocking method that loops internally until the client is connected.
Why would you use it in a loop?

There is a different method to connect async together with UA_Client_run_iterate.

o6-nbeier, 2026-10-22:
Closing for no activity.

closed 2026-10-22 by o6-nbeier