open62541/open62541 #6134: UA_Server_addDataSourceVariableNode When Data Not Available

MikeGranby, 2023-11-17:
Version 1.4 on Alpine Linux.

I'm trying to add a node to the data model that represents a register from a PLC. The PLC might be offline, or the source protocol might not yet have read the data. In those cases, there is no reliable value available, and so I'd ideally like to use something like the UA_STATUSCODE_UNCERTAININITIALVALUE to indicate to the client that the value I'm providing is not reliable. But if I don't return an actual value from the read method and the node is not writable, the node never gets inserted into the model. Is there some incantation required to get the node added, but without a valid value, or with the status code set? Or is this something that needs to be fixed? If so, I'm quite happy to have a look at it...

Log shows:

AddNode (ns=1;s=D1.P1): Using attributes for from the variable type failed with error code UncertainInitialValue

(This seems related to #4313, but it's not quite the same use case.)

V-Strassheim, 2023-11-20:
Hello Mike,
do you have any reason not to adjust the status code at server runtime, i.e. after the nodes have been defined?

At first glance, I would solve this with the "data source" callback.
This is explained in tutorial_server_datasource.c. See e.g. the writeCurrentTime method.

I have just adjusted the [return value](https://github.com/open62541/open62541/blob/08d2cbeca26683732721d29c768cfb535e20cc73/examples/tutorial_server_datasource.c#L116 ) when reading the example value and get the following displayed in the client: 
[image]

Regards,
Vasilij

MikeGranby, 2023-11-20:
I can make this work, but it's a bit of a kludge since how do I tell from within the read data hook whether it's being called during node addition or during actual runtime? It seems that `sessionId->namespaceIndex` is zero during the former, and non-zero during the latter, but is this a reliable and specified mechanism? Also, I'm adding nodes dynamically so I don't have the concept of startup and runtime as such, but I guess since everything appears to be single threaded and I'm looking for new nodes between calls to the `UA_Server_run_iterate` function, I could have a global variable to indicate the mode, or better still, a member variable in the data pointed to by `nodeContext`. Yeah, that'll work. It does seem a strange requirement to provide a value during startup, though, when that is precisely the point at which a value might not be available.

Alexk12, 2023-11-30:
I encountered the same problem and join your request. I have a datasource that is read-only and it is associated with a device value that may not yet available at startup.
Returning any status code other than success prevents adding a node to server's name space.

max65482, 2023-12-05:
#6164 proposes a solution to this issue.

closed 2023-12-13 by jpfr