open62541/open62541 #6143: Generated datatypes have namespaceIndex 0

matkonnerth, 2023-11-25:
## Description

In the tests for the nodesetcompiler this cmake macro is used, NAMESPACE_MAP is set.

ua_generate_nodeset_and_datatypes(
        NAME "tests-di"
        FILE_CSV "${UA_NODESET_DIR}/DI/Opc.Ua.Di.NodeIds.csv"
        FILE_BSD "${UA_NODESET_DIR}/DI/Opc.Ua.Di.Types.bsd"
        NAMESPACE_MAP "2:http://opcfoundation.org/UA/DI/"
        OUTPUT_DIR "${GENERATE_OUTPUT_DIR}"
        FILE_NS "${UA_NODESET_DIR}/DI/Opc.Ua.Di.NodeSet2.xml"
        INTERNAL
    )

But the generated datatypes (in src_generated/tests) have a typeId with namespaceIndex 0 instead of 2 as specified in namespace map.

## Background Information / Reproduction Steps

can be reproduced with master

Used CMake options:
BUILD_SHARED_LIBS:BOOL=ON
CLANG_FORMAT_EXE:FILEPATH=/usr/bin/clang-format
CMAKE_BUILD_TYPE:STRING=Debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
UA_ARCHITECTURE:STRING=posix
UA_BUILD_EXAMPLES:BOOL=OFF
UA_BUILD_TOOLS:BOOL=OFF
UA_BUILD_UNIT_TESTS:BOOL=ON
UA_ENABLE_AMALGAMATION:BOOL=OFF
UA_ENABLE_DA:BOOL=ON
UA_ENABLE_DIAGNOSTICS:BOOL=ON
UA_ENABLE_DISCOVERY:BOOL=ON
UA_ENABLE_ENCRYPTION:STRING=OFF
UA_ENABLE_ENCRYPTION_TPM2:STRING=OFF
UA_ENABLE_HISTORIZING:BOOL=ON
UA_ENABLE_JSON_ENCODING:BOOL=ON
UA_ENABLE_METHODCALLS:BOOL=ON
UA_ENABLE_NODESETLOADER:BOOL=OFF
UA_ENABLE_PUBSUB:BOOL=OFF
UA_ENABLE_PUBSUB_INFORMATIONMODEL:BOOL=OFF
UA_ENABLE_SUBSCRIPTIONS:BOOL=ON
UA_ENABLE_SUBSCRIPTIONS_EVENTS:BOOL=ON
UA_ENABLE_XML_ENCODING:BOOL=OFF
UA_FORCE_WERROR:BOOL=OFF
UA_LOGLEVEL:STRING=300
UA_MULTITHREADING:STRING=100
UA_NAMESPACE_ZERO:STRING=FULL
UA_NODESET_DIR:STRING=/home/matzy/git/open62541/deps/ua-nodeset

V-Strassheim, 2023-11-27:
Hi,
at a glance, it looks to me as if this has come in here:
https://github.com/open62541/open62541/pull/4916/commits/8e985d20341d2aaa0383fe96e433265acd470e48#diff-33bf9cdd39a74bf5c5f0ffdf917e3d12586b8715cc894d8b9b48b2d46a41d673

I would create a PR with a revert of the particular change, that works for me with the given examples.

NoelGraf, 2023-11-27:
With the nodesetinjector changes have been made. Thus, for example, the specification of the NamespaceMap is no longer necessary. Now it is checked at runtime which index the loaded nodeset has and is adapted accordingly in the types array.

V-Strassheim, 2023-11-27:
In this case, is it intended that the corresponding test types (e.g. DeviceHealthEnumeration) are now in namespace 0?

V-Strassheim, 2023-11-27:
Thanks, I got it now.

@matkonnerth do you have a problem with the current solution, where it is only adapted in runtime? Or is it only noticeable because it is different than before?

NoelGraf, 2023-11-27:
That is correct. I will take care of that.

NoelGraf, 2024-05-16:
https://github.com/open62541/open62541/pull/6214

closed 2024-05-16 by NoelGraf