open62541/open62541 #6152: OPC UA Server: Accept client application instance certificate based on trusted issuer list

alex0598, 2023-11-29:
## Description

I have a Open62541 Client Application and a separate  Open62541 Server Application.

The server application has a CA certificate in its trusted issuer list. The trusted certificate list is empty.
The client try to connect to the server application with a certificate signed by the CA certficate that is in the trusted issuers list of the server.

What needs to be changed to allow the client connection to the server based on the issuer list. I am not able to include the certificate of the client itself into the trusted list of the server ...

The server returns the error:

_"TCP 628 | SC 35 | Processing the message failed with error BadCertificateUntrusted"_

## Background Information / Reproduction Steps

**Client Connection:**
```
static bool ConnectToServer(UA_Client*& client, const char* endpoint) 
{
   // Create the config
   UA_ClientConfig* cc = UA_Client_getConfig(client);

   // Set to default config with no trust and issuer list
   UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey, NULL, 0, NULL, 0);
   UA_CertificateVerification_AcceptAll(&cc->certificateVerification);

   // Set securityMode and securityPolicyUri
   UA_ByteString_clear(&cc->securityPolicyUri);
   cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
   cc->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");

   // Set uri and client type
   UA_ApplicationDescription_clear(&cc->clientDescription);
   UA_String_clear(&cc->clientDescription.applicationUri);
   cc->clientDescription.applicationUri = UA_STRING_ALLOC(myUrn);
   cc->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;

   // Connect to the server
   if (UA_Client_connect(client, endpoint) != UA_STATUSCODE_GOOD) {
      return false;
   }

   return true;
}
```

**Server setup:**

```
...

// Update security policies
UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, " - I - Update Security Policies ...");
if (UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840, &certificate, &privateKey, trustList, trustListSize, issuerList, issuerListSize, revocationList, revocationListSize) != UA_STATUSCODE_GOOD) {
   UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, " - E - Failed to update Security Policies ...");
   return EXIT_FAILURE;
}

...
```

Used CMake options:

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-12-05:
Thanks for the complete issue report.
Are you using OpenSSL or mbedTLS?

alex0598, 2023-12-05:
Hi,
I am using open ssl.
Unfortunately I did not provide a valid revocation list as well as issuer list. Both are NULL in the above case. When I am providing an empty list the client can connect successfully.

jpfr, 2023-12-05:
Thanks.
I think it makes the most sense to treat empty list == NULL list.
We should change that @NoelGraf

NoelGraf, 2025-03-05:
There must be a certificate in the trust list. This can be the client certificate itself, the issuer certificate or any certificate in the certificate chain.

With the introduction of the GDS Push Management functionality, there has been a significant update in this section.

Everything should now work as expected. If you experience any problems, please reopen the issue.

closed 2025-03-05 by NoelGraf