open62541/open62541 #6159: Does nodeset_compiler work with Expanded Node Id?

nqduy35, 2023-12-04:
I have used nodeset_compiler a lot to generate OPC UA nodes in C code from NodeSet2. It is a very nice tool. Today, when I tried the tool with Expanded Node Id (nsu), it did not seem to work. Could you please confirm that nodeset_compiler does not support Expanded Node Id? If it is true, do you have any suggestion on how I can generate C code from a NodeSet2 including Expanded Node Id? Thank you in advance.  

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.9
 - [x] Other OPC UA SDKs used (client or server): UaExpert v1.7
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-12-05:
Where do you want to use the ExpandedNodeId?
As the value of a VariableNode?

nqduy35, 2023-12-05:
Thank @jpfr. I want to use nsu to replace ns in all UAObject and UAVariable nodes. Here is my scenario in detail: I have a custom nodeset including the resources of a robot (Objects and Variables). I can access these resources on my OPC UA server. In the future, there will be multiple robots like this one, and I received a suggestion to use ExpandedNodeId to manage all these nodes easier.

jpfr, 2023-12-05:
I think there is a possibility to define the Namespace URLs in a list at the top of the nodeset xml.
The namespace index of the NodeIds maps into that list.
The nodeset compiler takes that into account.

That is already very close to having ExpandedNodeIds where the Namespace Urls are listed individually in the NodeIds.

nqduy35, 2023-12-05:
Thank @jpfr again. I will try this approach.

closed 2023-12-05 by nqduy35