open62541/open62541 #6167: Client Does Not Honor SecurityLevel in Endpoint Selection

MikeGranby, 2023-12-06:
The client matches the first suitable endpoint in the list returned by the server without reference to the `SecurityLevel`. In the attached example, the server returns three endpoints, with security policies of None, Basic256, and Basic256Sha256, all of which are available to the client. These have increasing security levels, which means the client should pick Basic256Sha256, but it instead picks None. The only way to override this is to force an encrypted connection via `securityMode` in the config structure, but this required _a priori_ knowledge of the fact that the server will indeed support this. The fix is pretty trivial: iterate through the entire list, maintaining a record of the best policy found to date and allowing it to be overridden by better ones. I've coded a fix locally, and will submit if you like.

## Attachments

[files.zip](https://github.com/open62541/open62541/files/13583661/files.zip)

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.3.8-1478-g08d2cbeca
 - [X] Other OPC UA SDKs used (client or server): Ignition 8.1.4 OPC UA Server
 - [X] Operating system: Alpine 3.17
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [X] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-12-06:
Good point. You are right.

If you can provide code we make the change with attribution to you.
As a code snippet in this thread or as a GitHub pull request.

MikeGranby, 2023-12-06:
Will do. Let me tidy it up, and I'll submit.

MikeGranby, 2023-12-07:
Here you go. I moved the diagnostic print outside the loop to only show the selected results, and since this means I have to track the indices of the selections, I use these variables in place of the found flags. I also moved the check for the need to close the channel on a policy change since it would already be closed on a failure to find an endpoint or policy. I tried to keep the formatting as unchanged as possible, despite my editor's best efforts to the contrary!

````
/* Combination of UA_Client_getEndpointsInternal and getEndpoints */
static void
responseGetEndpoints(UA_Client *client, void *userdata,
    UA_UInt32 requestId, void *response) {
    UA_LOCK(&client->clientMutex);

    client->endpointsHandshake = false;

    UA_LOG_DEBUG(client->config.logging, UA_LOGCATEGORY_CLIENT,
                 "Received GetEndpointsResponse");

    UA_GetEndpointsResponse *resp = (UA_GetEndpointsResponse*)response;

    /* GetEndpoints not possible. Fail the connection */
    if(resp->responseHeader.serviceResult != UA_STATUSCODE_GOOD) {
        /* Fail the connection attempt if the SecureChannel is still connected.
         * If the SecureChannel is (intentionally or unintentionally) closed,
         * the connectStatus should come from there. */
        if(UA_SecureChannel_isConnected(&client->channel)) {
            client->connectStatus = resp->responseHeader.serviceResult;
            closeSecureChannel(client);
            UA_LOG_ERROR(client->config.logging, UA_LOGCATEGORY_CLIENT,
                         "GetEndpointRequest failed with error code %s",
                         UA_StatusCode_name(client->connectStatus));
        }

        UA_GetEndpointsResponse_clear(resp);
        UA_UNLOCK(&client->clientMutex);
        return;
    }

    /* Warn if the Endpoints look incomplete / don't match the EndpointUrl */
    Client_warnEndpointsResult(client, resp, &client->discoveryUrl);

    const size_t notFound = (size_t)-1;
    size_t bestEndpointIndex = notFound;
    size_t bestTokenIndex = notFound;
    UA_Byte bestEndpointLevel = 0;
    const UA_String binaryTransport = UA_STRING("http://opcfoundation.org/UA-Profile/"
                                                "Transport/uatcp-uasc-uabinary");

    /* Find a matching combination of Endpoint and UserTokenPolicy */
    for(size_t i = 0; i < resp->endpointsSize; ++i) {
        UA_EndpointDescription*endpoint = &resp->endpoints[i];

        /* Do we already have a better candidate? */
        if(endpoint->securityLevel < bestEndpointLevel)
            continue;

        /* Filter by the ApplicationURI if defined */
        if(client->config.applicationUri.length > 0 &&
           !UA_String_equal(&client->config.applicationUri,
                            &endpoint->server.applicationUri))
            continue;

        /* Look out for binary transport endpoints.
         * Note: Siemens returns empty ProfileUrl, we will accept it as binary. */
        if(endpoint->transportProfileUri.length != 0 &&
           !UA_String_equal(&endpoint->transportProfileUri, &binaryTransport))
            continue;

        /* Valid SecurityMode? */
        if(endpoint->securityMode < 1 || endpoint->securityMode > 3) {
            UA_LOG_INFO(client->config.logging, UA_LOGCATEGORY_CLIENT,
                        "Rejecting endpoint %lu: invalid security mode",
                        (long unsigned)i);
            continue;
[... 144 lines cut ...]
            "Selected endpoint %lu in URL %.*s with SecurityMode "
            "%s and SecurityPolicy %.*s",
            (long unsigned)bestEndpointIndex, (int)endpoint->endpointUrl.length,
            endpoint->endpointUrl.data, securityModeNames[endpoint->securityMode - 1],
            (int)endpoint->securityPolicyUri.length, endpoint->securityPolicyUri.data);

        /* Log the selected UserTokenPolicy */
        UA_LOG_INFO(client->config.logging, UA_LOGCATEGORY_CLIENT,
                    "Selected UserTokenPolicy %.*s with UserTokenType %s "
                    "and SecurityPolicy %.*s",
                    (int)tokenPolicy->policyId.length, tokenPolicy->policyId.data,
                    userTokenTypeNames[tokenPolicy->tokenType],
                    (int)securityPolicyUri->length, securityPolicyUri->data);
#endif

        /* Close the SecureChannel if a different SecurityPolicy is defined by the
         * Endpoint */
        if(client->config.endpoint.securityMode != client->channel.securityMode ||
           !UA_String_equal(&client->config.endpoint.securityPolicyUri,
                            &client->channel.securityPolicy->policyUri))
            closeSecureChannel(client);
    }
    UA_UNLOCK(&client->clientMutex);
}
````

MikeGranby, 2023-12-08:
I guess we also don't set SecurityLevel in the server's endpoint list, beyond 0 for not recommended and 1 for otherwise...

MikeGranby, 2023-12-09:
(I'm gonna create a PR for this for you.)

NoelGraf, 2023-12-11:
PR is merged.

PR #6171 (2023-12-10): fix(client): honor endpoint security policies
Pick endpoint based on highest security level rather than using first

Closes #6167
commit 9f37f89a (2023-12-10): fix(client): honor endpoint security policies

Pick endpoint based on highest security level rather than using first

Closes #6167
closed 2023-12-11 by NoelGraf