open62541/open62541 #6173: CA Chain could not be verified

alex0598, 2023-12-11:
## Description

I am currently testing the validation of an AIC that is signed by an intermediate CA.
This is the certificate chain: **RootCA -> IntermediateCA -> AIC.**

The Server is configured using _UA_ServerConfig_setDefaultWithSecurityPolicies_.
The TrustList contains: RootCA, IntermediateCA
The IssuerList contains: RootCA, IntermediateCA
The RevocationList contains: CRL from RootCA, CRL from IntermediateCA.

Unfortunatly if i want to connect to the Server with the above AIC, the Server always returns **BadCertificateUntrusted**.
The AIC does not contain the other certificates as a chain. The intermediate Cert does also not contain a chain.

What is going wrong here ? I thought if a add the Intermediate CA and Root CA cert to the issuer list the server should look up those certificates. I don't think it should be required to add the RootCA and IntermediateCA certs to the AIC at client call ?

[aic.txt](https://github.com/open62541/open62541/files/13630083/aic.txt)
[intermediate.cert.txt](https://github.com/open62541/open62541/files/13630086/intermediate.cert.txt)
[ca.cert.txt](https://github.com/open62541/open62541/files/13630087/ca.cert.txt)

Please find the PEM files attached. I had to convert them to pem because GitLab does not allow me to add a .DER file ...

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

NoelGraf, 2023-12-11:
I will take a closer look at the whole thing. Thank you for all of the information!

alex0598, 2023-12-11:
@NoelGraf Thanks you. I have attached the CA folders (Root CA and Intermediate CA) that was generated by openssl. In "my.zip" you can find a AIC signed with the intermediate CA. 

If you verify the certificates with openssl using the following openssl command using a combined.pem (Intermediate PEM string + AIC PEM string), openssl returns "OK"

`openssl -verify -CAfile "PKI/myCA/rootCA/cersts/ca.cert.pem" "combined.pem"   `

PS: Please note that the certificates attached were regenerated. The certificates in my first note were different to those attached now (But the error is the same :-)) 

[PKI.zip](https://github.com/open62541/open62541/files/13633548/PKI.zip)
[my.zip](https://github.com/open62541/open62541/files/13633569/my.zip)

NoelGraf, 2023-12-11:
I tried to reproduce the problems, but the scenarios worked for me. However, I had to create my own application certificate because the private key was not in the "my" folder. Maybe you can add it and I can try again.

What version of the open62541 stack are you using?

NoelGraf, 2023-12-11:
@alex0598 The download link to the zip does not work.

alex0598, 2023-12-11:
[my.zip](https://github.com/open62541/open62541/files/13636795/my.zip)
@NoelGraf Does this work ?

alex0598, 2023-12-11:
@NoelGraf I have now updated open62541 to v1.4.0 rc1.

The server is created as follows:
```
 // Update the servers config
 if (UA_ServerConfig_setDefaultWithSecurityPolicies(config, 4840, certStore.certificate, certStore.privateKey, trustList, trustListSize, issuerList, issuerListSize, revocationList, revocationListSize) != UA_STATUSCODE_GOOD) {
    UA_LOG_ERROR(UA_Log_Stdout, UA_LOGCATEGORY_CLIENT, " - E - Failed to update Security Policies ...");
    return false;
 }
```

----

The client is created as follows:

```
// Set to default config with no trust and issuer list
UA_ClientConfig_setDefaultEncryption(cc, *this->certStore->certificate, *this->certStore->privateKey, NULL, 0, NULL, 0);
UA_CertificateVerification_AcceptAll(&cc->certificateVerification);

// Set securityMode and securityPolicyUri
UA_ByteString_clear(&cc->securityPolicyUri);
cc->securityMode = UA_MESSAGESECURITYMODE_SIGNANDENCRYPT;
cc->securityPolicyUri = UA_String_fromChars("http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256");

// Set uri and client type
UA_ApplicationDescription_clear(&cc->clientDescription);
UA_String_clear(&cc->clientDescription.applicationUri);
cc->clientDescription.applicationUri = UA_String_fromChars(this->certStore->applicationUri);
cc->clientDescription.applicationType = UA_APPLICATIONTYPE_CLIENT;

// Connect to the server
if (UA_Client_connect(client, this->opcUaEndpoint) != UA_STATUSCODE_GOOD) {
   return false;
}
```

---

I have used the following config:

**Config1:**
**IssuerList:** ca.cert.der, intermediate.cert.der
**TrustList:** ca.cert.der, intermediate.cert.der
**RevocationList:** ca.crl, intermediate.crl

**Config2:**
**IssuerList:** intermediate.cert.der
**TrustList:** ca.cert.der
**RevocationList:** ca.crl, intermediate.crl

NoelGraf, 2023-12-12:
I have tried the 'my' certificate and get the error BADCERTIFICATEINVALID caused by an openssl X509_V_ERR_AKID_ISSUER_SERIAL_MISMATCH error. There seems to be something wrong with the attached certificates.

Also, I have only tested this on a Linux system. It would be interesting to see if you can confirm that the verification works on Linux.
There are sometimes differences in implementation between Linux and Windows.

Also, it would be good if you could attach another folder containing all certificates and keys (root ca, intermediate ca, my) that belong together.

alex0598, 2023-12-12:
@NoelGraf Please find the certificates attached. I have also attached a PEM file containing the intermediate CA file and the my CA file.

With openSSL on windows i have verified the certificate with the following command:

`openssl verify -verbose -CAfile ca.cert.pem "intermediate+my.txt"
[Zertifikate_open62541.zip](https://github.com/open62541/open62541/files/13647233/Zertifikate_open62541.zip)
`

alex0598, 2023-12-15:
The authority key information of my AIC certificate was invalid.

closed 2023-12-15 by alex0598