open62541/open62541 #6177: Work Around Broken OPC UA Foundation Server Code re Channel Sequence Numbers

MikeGranby, 2023-12-13:
## Description

I'm somewhat ambivalent about this one, but I'm going to capture it so that if it's rejected, there's at least a record:

The Red Lion OPC UA server uses the now unsupported ANSI C stack from the foundation. This contains an erroneous check on the channel sequence number that results in it reporting `OpcUa_BadSequenceNumberInvalid` if the sequence number from the client is greater than 1024 when the server starts up. We can recover from this condition by resetting the channel sequence number to zero if and only if that specific error code is returned in an ERR frame. This breaks what the spec says about not wrapping around until some big number, but then again, sequence numbers are not persistent and reset on restart anyway, so this is no different than that. If this were just Red Lion, I'd probably say they should fix it, but given this comes from the foundation's code, I suspect it's out there in other places, and we should thus Postel it.

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.3.8-1478-g08d2cbeca
 - [X] Other OPC UA SDKs used (client or server): Red Lion Crimson 3.x OPC UA Server
 - [X] Operating system: Alpine 3.17
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-12-13:
I checked the code, our client begins the SequenceNumber with zero at startup.
You say "if the sequence number from the client is greater than 1024 when the server starts up".
I don't understand how this can happen then.

We have a way of configuring workarounds in our stack. See here:
https://github.com/open62541/open62541/blob/cc5c7da6a11bf0fe012dbcd5af48a4f1d9fc6297/include/open62541/server.h#L112
https://github.com/open62541/open62541/blob/cc5c7da6a11bf0fe012dbcd5af48a4f1d9fc6297/include/open62541/common.h#L145

You could add a workaround that can be enabled/disabled in the config.
The default should be to interoperate with as many servers/clients as possible.

MikeGranby, 2023-12-13:
Sorry. I was not clear. Our client is running and is happily connected to the server with a sequence number of say 2000. The foundation-based server restarts, and now does not have any memory of its last sequence number. We reconnect with say 2001, and the server code performs the following check:

````
    /* The SequenceNumber shall also monotonically increase for all Messages
       and shall not wrap around until it is greater than 4 294 966 271
       (UInt32.MaxValue - 1 024). The first number after the wrap around
       shall be less than 1 024. */
    if(!((a_uSequenceNumber == a_pSecureChannel->uLastSequenceNumberRcvd + 1) ||
         ((a_uSequenceNumber < 1024) &&
          (a_pSecureChannel->uLastSequenceNumberRcvd > 4294966271u))))
    {
        OpcUa_GotoErrorWithStatus(OpcUa_BadSequenceNumberInvalid);
    }
````

This fails, and the server rejects our connection.

jpfr, 2023-12-13:
Hmm. Reconnecting implies a new TCP connection and a new SecureChannel.
Our client should reset the sequence number for the reconnect then…

MikeGranby, 2023-12-13:
Let me get a log and capture. I should have done that to begin with. Apologies! :)

MikeGranby, 2023-12-14:
The archive attached has a log and capture which seem to indicate that the sequence number is not being reset upon establishment of a new secure channel. Packets 45 _et seq_ show the establishment of a new channel after I forced a disconnect, from the server device, with packet 53 containing the `OpenSecureChannel` request with a sequence of 7. I cannot see anything in the code that explicitly resets the sequence number, beyond whatever zeroing goes on when a new channel object is allocated. And I'm not seeing that being called except on creation or clearing of the client object. Further, I'm not convinced (based on my admittedly rather old copy) that the spec requires anything beyond the sequence number to increase monotonically and meet certain wrapping constraints.

[files.zip](https://github.com/open62541/open62541/files/13679250/files.zip)

MikeGranby, 2023-12-22:
Just saw this. I'm still queasy about the whole sequence number thing. My old copy of the spec, as I noted, seems to talk about sequence numbers continuing across reconnected and TokenId re-issues.

o6-nbeier, 2026-10-20:
Thank you for opening this issue.

The sequence numbers are now reset when a SecureChannel is closed (see #6180), so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier