open62541/open62541 #6178: Map namespace index of generated types and loaded nodeset?

lukasberbuer, 2023-12-14:
How should a namespace be mapped if the nodeset is loaded at runtime with `UA_Server_loadNodeset` and the types are generated at compile-time?

Let's take the DI nodeset as an example: https://github.com/OPCFoundation/UA-Nodeset/tree/latest/DI
- the nodeset [Opc.Ua.Di.NodeSet2.xml](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/DI/Opc.Ua.Di.NodeSet2.xml) is loaded during runtime with the `UA_Server_loadNodeset` function. The next free namespace index will be the namespace index of the loaded nodeset (correct?).
- the custom types [Opc.Ua.Di.Types.bsd](https://github.com/OPCFoundation/UA-Nodeset/blob/latest/DI/Opc.Ua.Di.Types.bsd) can only be generated with the nodeset compiler during compilation (e.g. with the CMake function `ua_generate_datatypes`). But the namespace index has to be known in advance because it's hardcoded in the `UA_DataType` of each type for both the `typeId` and the `binaryEncodingId`.

```c
struct UA_DataType {
#ifdef UA_ENABLE_TYPEDESCRIPTION
    const char *typeName;
#endif
    UA_NodeId typeId;           /* The nodeid of the type */
    UA_NodeId binaryEncodingId; /* NodeId of datatype when encoded as binary */
    // ...
};
```

What is recommended way of doing this?

## Is ExpandedNodeId the solution?

If both the `typeId` and `binaryEncodingId` were of type `UA_ExpandedNodeId`, the `namespaceUri` could be provided instead of the namespace index and resolved during runtime. **Should `typeId` and `binaryEncodingId` of type `UA_ExpandedNodeId`?**

### `types_di_generated.h`
```c
/**
 * TransferResultErrorDataType
 * ^^^^^^^^^^^^^^^^^^^^^^^^^^^
 */
typedef struct {
    UA_Int32 status;
    UA_DiagnosticInfo diagnostics;
} UA_TransferResultErrorDataType;

#define UA_TYPES_DI_TRANSFERRESULTERRORDATATYPE 1
```

### `types_di_generated.c`
```c
const UA_DataType UA_TYPES_DI[UA_TYPES_DI_COUNT] = {
// ...
/* TransferResultErrorDataType */
{
    UA_TYPENAME("TransferResultErrorDataType") /* .typeName */
    {2, UA_NODEIDTYPE_NUMERIC, {15888LU}}, /* .typeId */            // -> UA_ExpandedNodeId{UA_NODEID_NUMERIC(0, 15888LU), "http://opcfoundation.org/UA/DI/", 0}
    {2, UA_NODEIDTYPE_NUMERIC, {15891LU}}, /* .binaryEncodingId */  // -> UA_ExpandedNodeId{UA_NODEID_NUMERIC(0, 15891LU), "http://opcfoundation.org/UA/DI/", 0}
    sizeof(UA_TransferResultErrorDataType), /* .memSize */
    UA_DATATYPEKIND_STRUCTURE, /* .typeKind */
    false, /* .pointerFree */
    false, /* .overlayable */
    2, /* .membersSize */
    TransferResultErrorDataType_members  /* .members */
},
// ...
};
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.4-rc1
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

ccvca, 2023-12-18:
I guess that the NodeSet Importer will also provide a data type definition for you, so there is no need to use the generated one:
https://github.com/open62541/open62541-nodeset-loader/blob/5e517fc78f5ff8c4cc56521321b983c0bef0a07b/backends/open62541/src/DataTypeImporter.c#L468

lukasberbuer, 2023-12-19:
Yes I agree @ccvca, the data type definition in the information model will be generated by the importer, but we still need the actual C `struct`/`enum` + `UA_<...>_TYPES` array with the corresponding `UA_DataType`s definitions. This information must be available at compile time and cannot be generated dynamically during runtime.

An this two-step approach of generating types and loading the information model during runtime is the crux for me. Both need the namespace index but the namespace index is depending on the loading order of the nodesets.

ccvca, 2023-12-19:
If I'm not mistaken, you should only need the struct/enum, as the UA_Datatype Definition is already known from the nodeset importer. Use `UA_Server_findDataType` to get the correct DataType definition. 
https://github.com/open62541/open62541/blob/d01f197153ca1cffecc3eb7b0d0027eec1ebd9e1/include/open62541/server.h#L1850-L1853

lukasberbuer, 2023-12-21:
You are right @ccva. I just modified the `server_nodeset_loader` example a bit to inspect the generated namespaces and the `customDataType`s:

```c
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/plugin/nodesetloader.h>

#include <signal.h>
#include <stdio.h>

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(int argc, const char *argv[]) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_Server *server = UA_Server_new();
    UA_ServerConfig_setDefault(UA_Server_getConfig(server));

    for (int cnt = 1; cnt < argc; cnt++) {
        if (UA_StatusCode_isBad(UA_Server_loadNodeset(server, argv[cnt], NULL))) {
            printf("Nodeset %s could not be loaded, exit\n", argv[cnt]);
            return EXIT_FAILURE;
        }
    }

    UA_StatusCode status = UA_STATUSCODE_GOOD;
    size_t namespaceIndex = 0;
    while (status == UA_STATUSCODE_GOOD) {
        UA_String namespaceUri;
        status = UA_Server_getNamespaceByIndex(server, namespaceIndex++, &namespaceUri);
        printf("Namespace %ld: %s\n", namespaceIndex, namespaceUri.data);
        UA_String_clear(&namespaceUri);
    }

    UA_ServerConfig* config = UA_Server_getConfig(server);
    const UA_DataTypeArray* customDataTypes = config->customDataTypes;
    while (customDataTypes != NULL) {
        for (size_t i = 0; i < customDataTypes->typesSize; ++i) {
            printf("%-30s namespaceIndex: %d, numericId: %d\n",
                customDataTypes->types[i].typeName,
                customDataTypes->types[i].typeId.namespaceIndex,
                customDataTypes->types[i].typeId.identifier.numeric
            );
        }
        customDataTypes = customDataTypes->next;
    }

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);

    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```

Output with `Opc.Ua.Di.NodeSet2.xml`:
```
Namespace 1: http://opcfoundation.org/UA/
Namespace 2: urn:open62541.server.application
Namespace 3: http://opcfoundation.org/UA/DI/
Namespace 4: (null)
UpdateBehavior                 namespaceIndex: 2, numericId: 333
ParameterResultDataType        namespaceIndex: 2, numericId: 6525
FetchResultDataType            namespaceIndex: 2, numericId: 6522
SoftwareVersionFileType        namespaceIndex: 2, numericId: 331
DeviceHealthEnumeration        namespaceIndex: 2, numericId: 6244
TransferResultDataDataType     namespaceIndex: 2, numericId: 15889
TransferResultErrorDataType    namespaceIndex: 2, numericId: 15888
```

So if custom data types are generated, only the `namespaceUri` and the (numeric) identifier `numericIdentifier` is known. To resolve the `UA_DataType`, following steps are necessary:
1. Get `namespaceIndex` of `namespaceUri`:

    ```c
    size_t namespaceIndex;
    UA_Server_getNamespaceByName(server, &namespaceUri, &namespaceIndex);
    UA_NodeId typeId = UA_NODEID_NUMERIC(namespaceIndex, numericIdentifier);
    ```

2. Get `UA_DataType` of `typeId`: 

    ```c
    const UA_DataType* dataType = UA_Server_findDataType(server, &typeId);
    ```

## Remaining questions/issues

1. How to resolve the `UA_DataType` client-side? `UA_Client_findDataType` won't work because it's not populated by the nodeset loader.
2. How to make sure that the generated `UA_DataType` matches the generated memory layout, e.g. of a `struct`?
3. `UA_Server_findDataType` will be way slower than direct access to the `UA_<...>_TYPES` array... any possible improvements?

ccvca, 2023-12-21:
> 3. `UA_Server_findDataType` will be way slower than direct access to the `UA_<...>_TYPES` array... any possible improvements?
In case you need these types often, just query them once and store / cache the pointers in an Array/Struct.