open62541/open62541 #6179: Using non-anonymous logins with no certifications to connect server apears errro

suiyue9527, 2023-12-15:
## Description
when i use non-anonymous logins witn no certification to connect server，the client shows error of  "BadSecurityModeInsufficient".
but i add the certification to the server and client receive it with non-anonymous logins,the error disapears.

## Background Information / Reproduction Steps
error like this:
[image]

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): ver:1.3.4
 - [ ] Other OPC UA SDKs used (client or server):  non
 - [ ] Operating system: linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2023-12-16:
Without a certificate no encryption is possible.
And I think you are prevented a password that is not encrypted.
We can make the behavior configurable on our end.

suiyue9527, 2023-12-18:
@jpfr maybe it can not work without a certificate,but i do not have evidence to support this.thanks for your replay.

closed 2026-10-20 by o6-nbeier