open62541/open62541 #6195: open62541.c:73504: UA_NodeMap_releaseNode: Assertion `entry->refCount > 0' failed.

aiweed, 2023-12-26:
问题：
open62541.c:73504: UA_NodeMap_releaseNode: Assertion `entry->refCount > 0' failed. 一个断言错误，程序直接退出，1.3.6版本，1.3.7版本和1.3.9版本都会出现

1、在一个单核的ARM  CPU上程序一直可以正常运行，放到双核的ARM   CPU运行一断时间就会出现上边的问题
2、具体情况是我从其它设备读取的数据通过UA_Server_writeValue写入node，我在所有引用到server的位置都加了全局锁

void Opcua::opcuaWriteValue(OpcuaNode *node, bool value)
{
    UA_Boolean myBool = value;
    UA_Variant myVar;
    UA_Variant_init(&myVar);
    UA_Variant_setScalar(&myVar, &myBool, &UA_TYPES[UA_TYPES_BOOLEAN]);

    serverMutex.lock();
    UA_Server_writeValue(server, node->nodeId, myVar);
    serverMutex.unlock();
}
3、编译 cmake -DUA_ENABLE_AMALGAMATION=ON  -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_SUBSCRIPTIONS=ON -DUA_ENABLE_SUBSCRIPTIONS_EVENTS=ON ..

Please provide the following information:

 - [ ] 1.3.9，1.3.7,1.3.9
 - [* ] Other OPC UA SDKs used (client or server): 
 - [ ] linux嵌入式，buildrootfs:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

谢谢

jpfr, 2023-12-28:
I cannot reproduce on x64.
Do you have the possibility to run the code with Valgrind to see if other memory errors occur before this?

o6-nbeier, 2026-10-20:
We were not able to reproduce this. If you still encounter this issue, please open a new issue with up to date information.

Closing for now.

closed 2026-10-20 by o6-nbeier