open62541/open62541 #6196: how to access OPC UA servers on local devices from OPC UA clients on remote cloud?

xibeisiber, 2023-12-26:
## Description

Hi,
My OPC UA servers and clients are not in the local network: the clients are hosted on remote cloud. How can the client access the servers considering local devices have no public domain.

It seems one possible way is to convert OPC UA messages to MQTT messages and then deliver them to the cloud. 

Is there a way to skip MQTT?
Thanks!
Jiying Jia

jpfr, 2023-12-28:
Either you publish over PubSub/MQTT.
Or you use the reverse connect feature to connect to the outside.
Instead of connecting to the inside.

jpfr, 2023-12-28:
Reverse connect will become available on the 1.4 release.

xibeisiber, 2023-12-30:
> Reverse connect will become available on the 1.4 release.

wow, thanks, looking forward to that

closed 2023-12-30 by xibeisiber