open62541/open62541 #6201: adding and removing nodes at runtime

Babu8008, 2024-01-03:
Hi Team,

I am new to this OPC UA and i am using open62541, I need to create a dynamic information model of adding and removing of nodes at runtime. it having one ObjectTypeNode and inside that objectTypeNode having four variable nodes of String type or int type. 

information model is looks like 
[image]

How to proceed this any papers are published for this to implement?

Please give some suggetions to implement the above requirement in  Embedded devices like stm32 development board in C-language.

Thanks in advance!

jpfr, 2024-01-04:
You want to add nodes locally or over the network?

Babu8008, 2024-01-04:
Hi jpfr,
Thanks for your replay.

I need to add nodes locally.

jpfr, 2024-01-04:
Here is the documentation for the server API: https://www.open62541.org/doc/master/server.html
There are lots of examples in the /examples folder.

Babu8008, 2024-01-04:
Hi jpfr,
Thanks for your replay.

If i need to add the nodes or remove the nodes over the network means what i need to follow.
any examples or present?

please share the any information regarding this implementation.