open62541/open62541 #6203: plugins/include/open62541/client_config_default.h broken with encryption support

JohannesKauffmann, 2024-01-03:
## Description
`plugins/include/open62541/client_config_default.h` has an `#ifdef UA_ENABLE_ENCRYPTION` to enable/disable the `UA_ClientConfig_setDefaultEncryption` function. However, configuring with `-DUA_ENABLE_ENCRYPTION_MBEDTLS=ON` does not enable the `UA_ENABLE_ENCRYPTION` define.

The symptoms of this can be seen when compiling `examples/pubsub/sks/pubsub_publish_encrypted_sks.c`, as this will give the following warning:
```
/open62541/examples/pubsub/sks/pubsub_publish_encrypted_sks.c:178:5: warning: implicit declaration of function ‘UA_ClientConfig_setDefaultEncryption’; did you mean ‘UA_ClientConfig_setDefault’? [-Wimplicit-function-declaration]
  178 |     UA_ClientConfig_setDefaultEncryption(cc, certificate, privateKey, 0, 0, NULL, 0);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     UA_ClientConfig_setDefault
/open62541/examples/pubsub/sks/pubsub_publish_encrypted_sks.c:178:5: warning: nested extern declaration of ‘UA_ClientConfig_setDefaultEncryption’ [-Wnested-externs]
```

## Background Information / Reproduction Steps
Build current master with:
```
CFLAGS="-Wstrict-prototypes" cmake .. -DUA_BUILD_EXAMPLES=ON -DUA_BUILD_TOOLS=ON -DUA_BUILD_UNIT_TESTS=ON -DUA_ENABLE_XML_ENCODING=ON -DUA_ENABLE_NODESETLOADER=ON -DUA_ENABLE_DISCOVERY_MULTICAST=ON -DUA_ENABLE_ENCRYPTION_MBEDTLS=ON -DUA_ENABLE_QUERY=ON -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_ENCRYPTION=ON -DUA_ENABLE_PUBSUB_SKS=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_PUBSUB_MONITORING=ON -DUA_ENABLE_MQTT=ON
```

JohannesKauffmann, 2024-01-03:
Actually, this was my own fault, for using `-DUA_ENABLE_ENCRYPTION_MBEDTLS=ON` instead of `-DUA_ENABLE_ENCRYPTION=MBEDTLS`. With the latter, encryption support is enabled and everything works as expected.

Malick31, 2024-05-16:
I have a question did you install MBEDTLS before using the cmake ?