open62541/open62541 #6209: Subscription publishing interval for variable that changes every 1s.

hmaiedEtic, 2024-01-04:
## Description

Hello,
I have a boolean variable that switches values every second (0-1) on my server. My OPC UA client subscribes to this variable and receives two values simultaneously:

OPC client: The value of variable BOOL has changed! New value: 1
OPC client: The value of variable BOOL has changed! New value: 0

I fixed "request.requestedPublishingInterval" to 250ms, and I used the default value for "request.requestedParameters.samplingInterval" (which is 250ms ). but i still recieves two values at the same time every second.

Any ideas or suggestions are welcome.
Thanks.

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.2.0
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: linux
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached 300
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-01-04:
starting points:

Check the returned actual sampling time and publishing time.

Update to the latest 1.3 stable release.

hmaiedEtic, 2024-01-05:
the returned sampling time is 250ms and publishing time is 250ms too. but i recieve two values every ~2s : 

```
[2024-01-05 10:34:19.853 (UTC+0100)] debug/channel      Connection 5 | SecureChannel 2 | Sending request with RequestId 16 of type PublishRequest
opc client: The value of var BOOL has changed! new value : 1
[2024-01-05 10:34:21.855 (UTC+0100)] debug/channel      Connection 5 | SecureChannel 2 | Sending request with RequestId 17 of type PublishRequest
opc client: The value of var BOOL has changed! new value : 0
[2024-01-05 10:34:21.856 (UTC+0100)] debug/channel      Connection 5 | SecureChannel 2 | Sending request with RequestId 18 of type PublishRequest
```

pfirsich, 2024-10-01:
Do you call `UA_Client_run_iterate` frequently enough? You will only get subscription callbacks, when you update the open62541 event loop.

closed 2024-11-13 by hmaiedEtic