open62541/open62541 #6211: Need instructions for generating single-file .c .h files.

davenmccalla-diagraph, 2024-01-04:
I have a project that using version 1.2.2 of open62541.c and open62541.h files.

I'd like to upgrade to version 1.3.9. How can I generate version 1.3.9 open62541.c and open62541.h of these files?

Here's what I tried:
```
git clone https://github.com/open62541/open62541.git
cd open62541
git checkout pack/v1.3.9
cmake -DUA_ENABLE_AMALGAMATION=TRUE

```
It complained that it couldn't find nmake and CMAKE_C_COMPILER not set, but I'm not trying to compile it, I just need the open62541.c and open62541.h generated.

Please help.

jpfr, 2024-01-04:
True, you don’t need a C compiler.
But we haven’t special-cased that.
CMake is for .. building C after all.

you could just install gcc and make once.

Version 1.3.6 has a pre-amalgamated build on GitHub.
I will make sure we add that for more releases…
https://github.com/open62541/open62541/releases/tag/v1.3.6

davenmccalla-diagraph, 2024-01-05:
It turns out I had to install at least one C compiler in Visual Studio to get nmake. It looks like the above command ran without errors, but I don't see the single-files anywhere.

**Do I have the correct branch checked out?** See above where I checked out branch "pack/v1.3.9".
**Do I have to run make after cmake to generate files?** When I do, it complains about missing separator.

davenmccalla-diagraph, 2024-01-05:
I figured it out. Here are some instructions for anyone else that needs to generate single-files:

- Download and extract the following zip file: https://github.com/open62541/open62541/archive/refs/tags/v1.3.9.zip
```
cd open62541-1.3.9/
mkdir build
cd build
cmake -DUA_ENABLE_AMALGAMATION=TRUE .. -G "Visual Studio 17 2022"
```
- Open the **open62541.sln** file in the build directory using Visual Studio.
- Select "Build" then "Build Solution" and the single-files will be generated and placed in the build directory.

closed 2024-01-06 by davenmccalla-diagraph