open62541/open62541 #6212: Writing to OPC UA server nodes from an MQTT Client

tedwad, 2024-01-05:
## Description

I have some basic questions about the interaction between OPC UA PubSub and MQTT. I have created an OPC UA server using a custom information model. I also have data being published as a list of strings to a remote Mosquitto broker at test.mosquitto.org:1883. I want to parse through the data from the MQTT broker and publish the values into specific nodes of the OPC UA server. In this way, my MQTT broker would act as a publisher and my OPC UA server as a subscriber. I looked through the open62541 documentation, but I have not been able to figure out how to get data from the broker into the nodes of my information model. Is this approach feasible with the open62541 library, or am I misunderstanding how OPC UA PubSub works? 

## Background Information / Reproduction Steps
I tried combining the steps from the tutorial_pubsub_mqtt_subscribe.c example and the custom information model guide from the documentation, and ended up with the following main.c file:

```
/* This work is licensed under a Creative Commons CCZero 1.0 Universal License.
 * See http://creativecommons.org/publicdomain/zero/1.0/ for more information. */

#include <open62541/plugin/log_stdout.h>
#include <open62541/server.h>
#include <open62541/server_config_default.h>
#include <open62541/server_pubsub.h>
#include <open62541/plugin/securitypolicy_default.h>

/* Files example_namespace.h and example_namespace.c are created from server_nodeset.xml in the
 * /src_generated directory by CMake */
#include "../open62541/build/src_generated/open62541/namespace0_generated.h"
#include "build/src_generated/di_nodeids.h"
#include "build/src_generated/namespace_di_generated.h"
#include "build/src_generated/rob_nodeids.h"
#include "build/src_generated/namespace_rob_generated.h"
#include "build/src_generated/ff_nodeids.h"
#include "build/src_generated/namespace_ff_generated.h"

#include <signal.h>
#include <stdio.h>
#include <stdlib.h>

#define CONNECTION_NAME               "MQTT Subscriber Connection"
#define TRANSPORT_PROFILE_URI_UADP    "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt-uadp"
#define TRANSPORT_PROFILE_URI_JSON    "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt-json"
#define MQTT_CLIENT_ID                "TESTCLIENTPUBSUBMQTTSUBSCRIBE"
#define CONNECTIONOPTION_NAME         "mqttClientId"
#define SUBSCRIBER_TOPIC              "customTopic"
#define SUBSCRIBER_METADATAQUEUENAME  "MetaDataTopic"
#define SUBSCRIBER_METADATAUPDATETIME 0
#define BROKER_ADDRESS_URL            "test.mosquitto.org:1883"

static UA_Boolean useJson = false;

UA_NodeId connectionIdent;
UA_NodeId subscribedDataSetIdent;
UA_NodeId readerGroupIdent;

UA_DataSetReaderConfig readerConfig;

static void fillTestDataSetMetaData(UA_DataSetMetaDataType *pMetaData);

static UA_StatusCode
addPubSubConnection(UA_Server *server, char *addressUrl) {
    UA_StatusCode retval = UA_STATUSCODE_GOOD;
    /* Details about the connection configuration and handling are located
     * in the pubsub connection tutorial */
    UA_PubSubConnectionConfig connectionConfig;
    memset(&connectionConfig, 0, sizeof(connectionConfig));
    connectionConfig.name = UA_STRING(CONNECTION_NAME);
    if (useJson) {
        connectionConfig.transportProfileUri = UA_STRING(TRANSPORT_PROFILE_URI_JSON);
    } else {
        connectionConfig.transportProfileUri = UA_STRING(TRANSPORT_PROFILE_URI_UADP);
    }
    connectionConfig.enabled = UA_TRUE;

    /* configure address of the mqtt broker (local on default port) */
    UA_NetworkAddressUrlDataType networkAddressUrl = {UA_STRING_NULL , UA_STRING(addressUrl)};
    UA_Variant_setScalar(&connectionConfig.address, &networkAddressUrl,
                         &UA_TYPES[UA_TYPES_NETWORKADDRESSURLDATATYPE]);
[... 390 lines cut ...]
```

Although the file compiles fine, I end up with the following output:

[image]

I am pretty inexperieced with C, so I am not sure if I am approaching this problem correctly. I would very much appreciate if someone could point me in the right direction if there is a better way to do this!

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo  -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON ..
```

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-01-05:
You should add MQTT support to the build flags.
This all works…. But: the messages on MQTT need to be OPC UA PubSub messages, not just any MQTT.
If you look at our internal MQTT API in /include/open62541/plugins/eventloop.h you might have an easier time spinning up your own MQTT client, atttached to the EventLoop of the OPC UA server, and process the incoming data in your own callback method. There you can write the data to the nodes.

tedwad, 2024-01-06:
Would you suggest using the mqtt-c library to create the MQTT client and attaching it to the EventLoop of the server? Or is creating the MQTT client possible using only the definitions from including the eventloop.h header file?

jpfr, 2024-01-07:
The MQTT-C library was wrapped to be used for our PubSub subsystem.
You only need the definitions from eventloop.h to use it.

jpfr, 2024-01-07:
This is the internal creation of a MQTT subscriber for PubSub:
https://github.com/open62541/open62541/blob/aa98741b4f9dacd89bceb36cbd5846b1a40e8b1f/src/pubsub/ua_pubsub_eventloop.c#L892

tedwad, 2024-01-09:
I managed to get my application working by following your suggestion, thank you for all the help!

closed 2024-01-09 by tedwad