open62541/open62541 #6213: dynamically update the information model in open62541

Babu8008, 2024-01-05:
Hi Team,

I am using OPC UA open62541 stack. I created opc ua server, statically i can able to create the information model having multiple ObjectNodes and inside each objectNode having 4 variable nodes.
[image]
Now i need to create dynamic information model at runtime that means my server will not stop if i load the XML file (or any other possibility ways) to the server it has to read that XML file and display the information model at Address space in UA expert client application.

Please suggest how to implement this feature.

if any examples present in OPC UA foundation, please suggest those examples also.

if it not works via XML file uploading to the server please suggest how to proceed with other possibilities 

Thanks in advance

I am expecting at runtime i need to modify the information model.

jpfr, 2024-01-05:
Have a look at the API:
AddNode, DeleteNode, AddReference, DeleteReference.
This API can also be used remotely.
If you want to upload XML, there is a node set-loader plugin that can be used to load node definitions from xml at runtime.

o6-nbeier, 2026-10-20:
Closing for now.

closed 2026-10-20 by o6-nbeier