open62541/open62541 #6219: Issues compiling using open62541.c/.h files

om327, 2024-01-08:
Hello, 

I'm probably doing something daft, but  I can't seem to build the simple server from your documentation -https://www.open62541.org/doc/master/tutorial_server_firststeps.html

I've followed your build instructions for Windows with the “amalgamation” option enabled and complied with CMake and Visual Studio 2022. 
```
git clone https://github.com/open62541/open62541.git
cd .\open62541\
git submodule update --init --recursive
mkdir build
cd .\build\
cmake -G "Visual Studio 17 2022" ..
cmake-gui (selected SHARED_LIBS and AMALGAMATION)
cmake --build . --config Release
```
This has produced the open62541.c/.h files, which I copied to a new folder and initially created myServer.c with the following:

```
#include <open62541/server.h>

int main(void) {
    UA_Server *server = UA_Server_new();
    UA_StatusCode retval = UA_Server_runUntilInterrupt(server);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```
I noticed that in this example, you're including a static typed ```<open62541/server.h>```, which I'm a little confused about since the guide suggests all we need are the open62541.c/.h. files in the same directory. When I try to build, I get an error that ```open62541/server.h can not be found```.

As these are in the same folder, my understanding is we should use double quotes ```" "``` rather than ```<>```, so I also tried the following.
```
#include "open62541"

int main(void) {
    UA_Server *server = UA_Server_new();
    UA_StatusCode retval = UA_Server_runUntilInterrupt(server);
    UA_Server_delete(server);
    return retval == UA_STATUSCODE_GOOD ? EXIT_SUCCESS : EXIT_FAILURE;
}
```
I have the msys2 toolchain installed and working with vs code. This should allow me to use the following: ```gcc -std=c99 myServer.c -lopen62541 -lws2_32 -o myServer.exe``` but I get the same error but this time ```open62541.h can not be found```?

The strange thing is that intelligence seems to work since hovering over the functions displays the relevant text.

[image]

I'm relatively new to c/c++ (my background is Python), so I might be doing something silly, but the instructions seem contradictory with the example referencing ```#include <open62541/server.h>```, which I understand would be the static typed and installed files for Visual Studio.

Any help would be fantastic!

kimim, 2024-01-08:
As you are using msys2 now, you can compile with cmake and ninja:
```
mkdir build
cd build
cmake -DUA_ENABLE_AMALGAMATION=ON ..
ninja
```
Then copy `open62541.[c|h]` to the same folder as `myServer.c`
```
gcc -std=c99 open62541.c myServer.c -lws2_32 -lIphlpapi -o myServer.exe
```
`-lIphlpapi`  is used for `if_nametoindex`

om327, 2024-01-08:
Thanks, Kimim!

I've tried your commands but couldn't get them to work. 

I installed Ninja through Python, as Chocolatey didn't work even after adding the ```choco/bin``` location where the ninja.exe was saved to my PATH.  

I replaced your ```ninja``` cmd with ```cmake -G Ninja``` from within the ```open62541``` folder after first building the files with ```cmake -DUA_ENABLE_AMALGAMATION=ON ..```. 

I then copied over the .h/.c filers and tried building the myServer.c again using  ```gcc -std=c99 open62541.c myServer.c -lws2_32 -lIphlpapi -o myServer.exe``` but this didn't work?

Could you tell me how to set up Visual Studio so I can ```#import <open62541/server.h>``` and others into a new cpp console solution and then build it?

jpfr, 2024-01-08:
The -lopen62541 is for system-wide lookup of an installed library. You should rather refer to open62541.c together with myServer.c.

the default is for gcc to look for header files in the same directory. If you want to look somewhere else, add the path with -I<header-path>.

kimim, 2024-01-20:
cmake and ninja work out of box from msys2. 
`pacman -S mingw-w64-ucrt-x86_64-cmake mingw-w64-ucrt-x86_64-cninja`