open62541/open62541 #6234: Basic256Sha256 security failed returns BadSecurityChecksFailed

Babu8008, 2024-01-17:
Hi Team,

I enabled security using certificates feature in open62541 and i generated keys for Basic256Sha256 security policy.
Now the issue is if i try to connect with Basic256Sha256 is Failed  but with same security cerificates i can able to connect with Aes128_Sha256_RsaOaep security policy. 
what might be the isuue? i am unable to sort out ,pleae help me to overcome this issue.

if i try to connect with Basic256Sha256 its showing like this
[image].

jpfr, 2024-01-17:
Please show the server-side log output.
That should be more informative.

Babu8008, 2024-01-18:
Hi ,
here i am attaching the logs from server side
[image]

jpfr, 2024-01-18:
Hmm. The logs are not very informative either.
If you are on the master or 1.4 branch we can improve the logging.

Or, if you have a debugger, you could set a breakpoint in the few locations where UA_STATUSCODE_BADSECURITYCHECKSFAILED gets thrown.

closed 2025-02-26 by andreasebner