open62541/open62541 #6237: Published json via MQTT is mixed up

MarcelArmgart, 2024-01-18:
## Description
Hi Team,

I am currently workling with the sources of v1.4 (#5e76d257fd0ab4bc2558bf5e24ba4d5f0bab0aec)
I enabled MQTT publish via json-encoding on a server. Therefore I used the example "tutorial_pubsub_mqtt_publish.c".
In the example I changed the line 50 from
`#define TRANSPORT_PROFILE_URI        "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt"`
to 
`#define TRANSPORT_PROFILE_URI        "http://opcfoundation.org/UA-Profile/Transport/pubsub-mqtt-json"`
as suggested in issue [#6129](https://github.com/open62541/open62541/pull/6129).

After that the example works perfectly and publishes the current server time in the expected interval.
But the json-string which is send to the mqtt broker is mixed up.

I subscribed the broker with MQTT Explorer and mosquitto_sub.exe via PowerShell this is what was send by the server:
`{null"MessageId":"ua-data","MessageType":,"PublisherId":2234,"Messages":[{62541"DataSetWriterId":2,"SequenceNumber":{"MajorVersion":464029280,"MinorVersion":464029280},"MetaDataVersion":"2024-01-18T12:46:00.289Z","Timestamp":0,"Status":,"Payload":{"Server localtime":{"Type":13,"Body":"2024-01-18T12:46:00.289Z"}}}]}`

Correct would be:
`{
    "MessageId":null,
    "MessageType":"ua-data",
    "PublisherId":2234,
    "Messages": [{
        "DataSetWriterId":62541,
        "SequenceNumber":2
        "MetaDataVersion":{
            "MajorVersion":464029280,
            "MinorVersion":464029280
        },
        "Timestamp":"2024-01-18T12:46:00.289Z",
        "Status":0,
        "Payload":{
            "Server localtime":{
                "Type":13,
                "Body":
                "2024-01-18T12:46:00.289Z"
            }
        }
    }]
}`

What am I doing wrong?

## Background Information / Reproduction Steps
Sources are buid from current v1.4 (Git-Hash: #5e76d257fd0ab4bc2558bf5e24ba4d5f0bab0aec)

Used CMake options:

```bash
cmake -DUA_NAMESPACE_ZERO=FULL -DBUILD_SHARED_LIBS=ON -DUA_ENABLE_ENCRYPTION=OPENSSL -DUA_ENABLE_PUBSUB=ON -DUA_ENABLE_PUBSUB_FILE_CONFIG=ON -DUA_ENABLE_PUBSUB_INFORMATIONMODEL=ON -DUA_ENABLE_JSON_ENCODING=ON -DUA_ENABLE_MQTT=ON
```

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 5e76d257fd0ab4bc2558bf5e24ba4d5f0bab0aec
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Windows 10
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-01-22:
That is really strange.
We have a unit test `check_pubsub_encoding_json` for exactly this.

Please run the unit test so we can pinpoint the problem better.
Also, which compiler are you using?

MarcelArmgart, 2024-01-23:
Ok I have built the unittests and your suggested test `check_pubsub_encoding_json` works fine. There were no errors.
The compiler was Visual Studio 2015 with VS 14.

MarcelArmgart, 2024-01-23:
So I finally found out why it was previously working on my machine.

The mixed up JSON message occurs, if the open62541.dll was compiled in RELEASE.
If the open62541.dll is compiled in DEBUG everything works fine.

So I will work with the DEBUG compiled file.

jpfr, 2024-01-23:
Still, that difference is unacceptable.
We will investigate that further.

MarcelArmgart, 2024-02-12:
Hello I finally found the problem and a solution for me and maybe for us all. The issue is located in file "ua_types_encoding_json.c".
In line 172 - 176 the function `writeJsonObjElm` is defined.

The issue than comes from line 175:
`return writeJsonKey(ctx, key) | encodeJsonJumpTable[type->typeKind](ctx, value, type);`
The bitwise combination these two function works in Debug mode fine.
First `writeJsonKey(ctx, key)` is called then `encodeJsonJumpTable[type->typeKind](ctx, value, type)` is called.

But in Release mode in Visual Studio 2015 the optimisation switches the order of these function calls.
First `encodeJsonJumpTable[type->typeKind](ctx, value, type)` is called then `writeJsonKey(ctx, key)` is called.

I fixed this issue for me by just spending an own line for each function call.
```status
writeJsonObjElm(CtxJson *ctx, const char *key,
                const void *value, const UA_DataType *type) {
    status ret = writeJsonKey(ctx, key);
    ret |= encodeJsonJumpTable[type->typeKind](ctx, value, type);
    return ret;
}

closed 2025-06-05 by jpfr

RolfKal, 2025-06-11:
Interesting find and I think a bug in Visual Studio 2015. If I interpret the currently accepted C Operator Precedence and Associativity list correctly. the bitwise OR operator is left to right associative. That means it SHOULD NOT change evaluation order like that.

RolfKal, 2025-06-11:
> [@RolfKal](https://github.com/RolfKal) No, [operator precedence](https://en.cppreference.com/w/c/language/operator_precedence.html) and [evaluation order](https://en.cppreference.com/w/c/language/eval_order.html) are two completely different things.

But its related to associativity which is also part of the operator precedence list, isn't it?