open62541/open62541 #6242: Several properties missing for ObjectTypes in reduced nodeset

cldhms, 2024-01-22:
## Description
CTT complains about several missing properties for some ObjectTypes when using the reduced nodeset.

Here are some examples:
ServerType (from [Opc.Ua.NodeSet2.Reduced.xml](https://github.com/open62541/open62541/blob/e1efda210cb8c8826f5e6d2c2cdb5b90a77ada1c/tools/schema/Opc.Ua.NodeSet2.Reduced.xml#L781)):
```
<UAObjectType NodeId="i=2004" BrowseName="ServerType">    
   <DisplayName>ServerType</DisplayName>
   <References>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=58</Reference>
   </References>
</UAObjectType>
```
For this type CTT expects the following properties:
ServerArray, NamespaceArray, UrisVersion, ServerStatus, ServiceLevel, Auditing, EstimatedReturnTime, LocalTime

OperationLimitsType (from [Opc.Ua.NodeSet2.Reduced.xml](https://github.com/open62541/open62541/blob/e1efda210cb8c8826f5e6d2c2cdb5b90a77ada1c/tools/schema/Opc.Ua.NodeSet2.Reduced.xml#L836)):
```
<UAObjectType NodeId="i=11564" BrowseName="OperationLimitsType">
   <DisplayName>OperationLimitsType</DisplayName>
   <References>
      <Reference ReferenceType="HasProperty">i=11565</Reference>
      <Reference ReferenceType="HasProperty">i=11567</Reference>
      <Reference ReferenceType="HasProperty">i=11569</Reference>
      <Reference ReferenceType="HasProperty">i=11570</Reference>
      <Reference ReferenceType="HasProperty">i=11571</Reference>
      <Reference ReferenceType="HasProperty">i=11572</Reference>
      <Reference ReferenceType="HasProperty">i=11573</Reference>
      <Reference ReferenceType="HasProperty">i=11574</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=61</Reference>
   </References>
</UAObjectType>
```
For this type CTT expects the listed properties above and additionally the following properties:
MaxNodesPerHistoryReadData, MaxNodesPerHistoryReadEvents, MaxNodesPerHistoryUpdateData, MaxNodesPerHistoryUpdateEvents

Example output from CTT:
`<ResultNode name="Error" timestamp="2024-01-15 12:54:57.020" testresult="0" description="OperationLimitsType -&gt; MaxNodesPerHistoryReadData ==&gt;[NOT FOUND] REQUIRED">`

I wasn't entirely sure if it was allowed to omit properties for known types or not so I asked a question over at the CTT Mantis ([for this related issue](https://mantis.opcfoundation.org/view.php?id=8655#c20668)) where I got the following reply:
_Types are not allowed to omit elements, no matter whether they are mandatory or optional. The types are always required to follow 100% the definition of the specification._

Should Opc.Ua.NodeSet2.Reduced.xml be updated to include all properties for the ObjectTypes included in it?

## Background Information / Reproduction Steps
Used CMake options:
UA_NAMESPACE_ZERO="REDUCED"

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): 1.3.7
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-01-22:
I'm not sure if the reduced nodeset can be ever used for a fully compliant server.
In order to reduce the time effort, I would recommend to use the v1.4-rc1 release candidate and the full nodeset.
A lot of effort went into the ongoing certification effort that targets 1.4.

cldhms, 2024-01-22:
Ok, that would explain things. According to the documentation "Reduced" should be CTT compliant so that may be in need of an update then.

From open62541 Documentation PDF file:
• REDUCED: Small namespace zero that passes the CTT

I plan to try 1.4 soon but I'm not sure how the full nodeset would affect memory footprint.