open62541/open62541 #6244: why clientHandle was revised when I call UA_Client_MonitoredItems_createDataChanges?

lixianshang, 2024-01-23:
## Description
I want to set MonitoredItem's clientHandle when I call UA_Client_MonitoredItems_createDataChanges, but I found that the value was revised before sending the request, can anybody explain for me what's the purpose? thanks!
```c
static UA_StatusCode
MonitoredItems_CreateData_prepare(UA_Client *client,
                                  const UA_CreateMonitoredItemsRequest *request,
                                  void **contexts, void **handlingCallbacks,
                                  UA_Client_DeleteMonitoredItemCallback *deleteCallbacks,
                                  MonitoredItems_CreateData *data) {
    /* Align arrays and copy over */
    UA_StatusCode retval = UA_STATUSCODE_BADOUTOFMEMORY;
    data->contexts = (void **)UA_malloc(sizeof(void *) * request->itemsToCreateSize);
    if(!data->contexts)
        goto cleanup;
    memcpy(data->contexts, contexts, request->itemsToCreateSize * sizeof(void *));

    data->deleteCallbacks = (UA_Client_DeleteMonitoredItemCallback *)
        UA_malloc(request->itemsToCreateSize * sizeof(UA_Client_DeleteMonitoredItemCallback));
    if(!data->deleteCallbacks)
        goto cleanup;
    memcpy(data->deleteCallbacks, deleteCallbacks,
           request->itemsToCreateSize * sizeof(UA_Client_DeleteMonitoredItemCallback));

    data->handlingCallbacks = (void **)
        UA_malloc(request->itemsToCreateSize * sizeof(void *));
    if(!data->handlingCallbacks)
        goto cleanup;
    memcpy(data->handlingCallbacks, handlingCallbacks,
           request->itemsToCreateSize * sizeof(void *));

    retval = UA_CreateMonitoredItemsRequest_copy(request, &data->request);
    if(retval != UA_STATUSCODE_GOOD)
        goto cleanup;

    /* Set the clientHandle */
    for(size_t i = 0; i < data->request.itemsToCreateSize; i++)
        data->request.itemsToCreate[i].requestedParameters.clientHandle =
            ++client->monitoredItemHandles;

    return UA_STATUSCODE_GOOD;

cleanup:
    MonitoredItems_CreateData_clear(client, data);
    return retval;
}
```
## Background Information / Reproduction Steps

## Checklist

Please provide the following information:

 - [x] open62541 Version (release number or git tag): v1.3.4
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2024-01-24:
Is this what you are referring to?
https://github.com/open62541/open62541/blob/16a182ea2e2fb872a0d0a826dc4b498f125ee697/src/client/ua_client_subscriptions.c#L496

The handle is set internally. So we can make sure there are no duplicate handles.
You are supposed to define a context (void) pointer for the MonitoredItems.
So you can disambiguate in the callback.

You can (of course) use some handle integer instead of the void pointer.

lixianshang, 2024-01-25:
Yes,I referred the file "ua_client_subscriptions.c". I found in the wireshark only ClientHandle was shown in MonitoredItemNotification, if the handle was set internally, I can't establish the relationship between the handle and nodeid when I analysing the wireshak message.

closed 2024-01-27 by lixianshang