open62541/open62541 #6245: Dead Lock in A&C example tutorial_server_alarms_conditions.c

LinuxZrc, 2024-01-23:
## Description
Hello,

I notived a deadlock issue during the testing example tutorial_server_alarms_conditions.c

open62541 server lost opc.tcp connetion with uaexpert client after method ConditionSourceObject->Condition 1->Enable call.

There may be a deadlock problem in alarms_condition example enable call :

```c
  UA_Server_enteringEnabledState->UA_LOCK(&server->serviceMutex)
  ->enteringEnabledStateCallback->UA_Server_writeObjectProperty_scalar->UA_UNLOCK(&server->serviceMutex)
```

## Background Information / Reproduction Steps

```c
tutorial_server_alarms_conditions.c : 347

static UA_StatusCode
enteringEnabledStateCallback(UA_Server *server, const UA_NodeId *condition) {
    UA_Boolean retain = true;
    return UA_Server_writeObjectProperty_scalar(server, *condition,
                                                UA_QUALIFIEDNAME(0, "Retain"),
                                                &retain,
                                                &UA_TYPES[UA_TYPES_BOOLEAN]);
}

ua_services_attribute.c : 2127

UA_StatusCode UA_EXPORT
UA_Server_writeObjectProperty_scalar(UA_Server *server, const UA_NodeId objectId,
                                     const UA_QualifiedName propertyName,
                                     const void *value, const UA_DataType *type) {
    UA_LOCK(&server->serviceMutex);
    UA_StatusCode retval = 
        writeObjectProperty_scalar(server, objectId, propertyName, value, type);
    UA_UNLOCK(&server->serviceMutex);
    return retval;
}

ua_subscription_alarms_conditions.c : 881

static UA_StatusCode
UA_Server_enteringEnabledState(UA_Server *server, const UA_NodeId *conditionId,
                               const UA_NodeId *conditionSource) {
    UA_LOCK(&server->serviceMutex);
    UA_StatusCode res = enteringEnabledState(server, conditionId, conditionSource);
    UA_UNLOCK(&server->serviceMutex);
    return res;
}
```

Used CMake options:

set(UA_NAMESPACE_ZERO "FULL")
set(UA_ENABLE_SUBSCRIPTIONS_ALARMS_CONDITIONS"EN")

```bash
cmake -DUA_NAMESPACE_ZERO=<YOUR_OPTION> <ANY_OTHER_OPTIONS> ..
```

## Checklist

Please provide the following information:

 - [X] open62541 Version (release number or git tag): v1.4.0-rc1-237-gd3e148ae6
 - [ ] Other OPC UA SDKs used (client or server): 
 - [X] Operating system: Arm-Linux-x64
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [X] Critical issue

V-Strassheim, 2024-01-24:
This seems to be a similar problem as here: https://github.com/open62541/open62541/issues/5351
I think the locks should be made more granular and released again before the callbacks, or really only lock the resources that need to be kept consistent. For the current implementation, reentrant/recursive mutexes that can be called nested by a thread would be simpler, but less pleasant fix.

@jpfr  / @andreasebner  Is it a special thing about the A&C module because there are more callbacks for certain states? Or could the nested locking also occur in other places?

A quick fix for the user would be to set UA_MULTITHREADING <100. However, depending on the application, this could have certain disadvantages.

jpfr, 2024-01-24:
Right. In our defense, A&C is still marked as experimental.
So it doesn't receive the same level of scrutiny and test coverage.

To get this right, whenever we call one of the callback methods, it needs to be wrapped into UA_UNLOCK and UA_LOCK.
You should be able to find all relevant by places by a search for `Callback(`.

giuseppedevito, 2024-04-04:
Same problem for me...and I can't do

UA_UNLOCK(&server->serviceLock)
.....
UA_LOCK(&server->serviceLock)

since server is an Opaque struct.

Compiler error:
error: invalid use of incomplete typedef ‘UA_Server’

What is the best way to unlock and than lock again server->serviceLock from a custom C file?

jpfr, 2024-04-04:
Please try again after #6381 was merged to 1.4.
You should never have to mess with the locks in your user code.

jpfr, 2024-04-05:
Please reopen if required.

closed 2024-04-05 by jpfr

LinuxZrc, 2024-04-07:
> Please try again after #6381 was merged to 1.4. You should never have to mess with the locks in your user code.

Thank you， the issue has been solved.