open62541/open62541 #6248: Questions about the TSN examples (broken for build ?)

jpanisbl, 2024-01-25:
Hello,

I have a couple of questions about the TSN examples in the `pubsub_realtime` directory.

I can read in `examples/CMakeLists.txt`:
`# TODO: The TSN examples need to be reworked for the EventLoop model`

I un-commented the related lines in `examples/CMakeLists.txt` to get them built. Then, for some of these examples:
- I had to add missing inclusions (`stdio.h`, `unistd.h`, `signal.h`).
- I had to comment some code, especially the code related to `UA_EthernetWriterGroupTransportDataType` because this structure is not defined.

I intended to use these examples to test pubsub XDP by following this document: `https://github.com/open62541/open62541/blob/master/examples/pubsub_realtime/README_pubsub_TSN.txt)`.

Questions:
1) Were these examples ever built/run/tested ?
2) Should I define `UA_EthernetWriterGroupTransportDataType` somewhere ?

Julien

TheRoslyak, 2024-02-07:
Support you. I have the same problem

TheRoslyak, 2024-02-08:
In general, I started to dig into it and found out the following. Version 1.3.9 works with TSN, but supposedly this specific type is not used. The examples pubsub_TSN_publisher_multiple_thread.c and pubsub_TSN_loopback_single_thread.c work (but there you need to change a couple of lines). Currently, the functionality of the function void *publisherETF(void *arg); is performed by the function void *pubApp(void *arg);. Now I will be redoing the example in a new way and for my task based on this tutorial: https://www.open62541.org/doc/1.3/pubsub_TSN_publisher.html. It would be nice to have some collaborative continuation. Otherwise, they messed up the library, but the thing is useful!!

jpanisbl, 2024-02-23:
Hi @TheRoslyak 
Sorry, I did not work on this topic any longer. Before that, I need to add XDP support in the linux driver we use.
Were you finally able to get the example work ?