open62541/open62541 #6251: Server rejects ActivateSession with NULL userIdentityToken.

MurzynJ, 2024-01-28:
Part 4 tells that NULL or empty userIdentityToken shall be interpreted as anonymous.

[image]

It worked before but it looks like it got broken during refactoring in commit a9d3787. Function Service_ActivateSession returns Bad_IdentityTokenInvalid in line 766.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): a6a9c7b

MurzynJ, 2024-11-27:
This is still not resolved. Some clients may not be able to connect to open62541 server if this is not fixed.

To reproduce the issue quickly, add few lines like this and connect from UA Expert with "None,Anonymous"

[image]

o6-nbeier, 2026-10-20:
Thank you for the report.

A null or empty token is now interpreted as anonymous (see 7eee5e7), so we are closing this as fixed.

closed 2026-10-20 by o6-nbeier