open62541/open62541 #6252: PubSub UDP - unicast subscriber not listening on all interfaces.

MurzynJ, 2024-01-28:
Part 14 defines behaviour of the UDP unicast subscribers as following.

[image]

The hostname part of the PubSubConnection's Address URL has no meaning in this case and is by definition set to a magic value 'localhost'. If not explicitly specified by NetworkInterface parameter, a subscriber shall listen on all interfaces.

With current implementation, the UDP connection is set to listen on loopback interface only.

## Checklist

Please provide the following information:

 - [ ] open62541 Version (release number or git tag): a6a9c7b

keba-estr, 2024-02-05:
Also discussed in issue #5915

jpfr, 2024-02-15:
This is addressed in PR #6282 for 1.4 and after.
It slipped through the unit tests, because there the messages originate from the local box.
Please verify #6802 and close this if fixed.

closed 2024-03-04 by MurzynJ