open62541/open62541 #6256: Allow custom definition of UA_FLOAT_IEEE754

m-obi, 2024-01-30:
I'm using an ESP32 and it's using the IEEE754 representation.
Therefore I'm defining the UA_FLOAT_IEEE754 with 1 in my CMakeLists.txt.

But this line of code is redefining the UA_FLOAT_IEEE754 with 0.
https://github.com/open62541/open62541/blame/cc806cb4719f6e392e7869b4a9e4d4a61d54ea3e/include/open62541/config.h.in#L621C23-L621C23

I changed the my block from
```#if defined(_WIN32)
# define UA_FLOAT_IEEE754 1
#elif defined(__i386__) || defined(__x86_64__) || defined(__amd64__) || \
    defined(__ia64__) || defined(__powerpc__) || defined(__sparc__) || \
    defined(__arm__)
# define UA_FLOAT_IEEE754 1
#elif defined(__STDC_IEC_559__)
# define UA_FLOAT_IEEE754 1
#else
# define UA_FLOAT_IEEE754 0
#endif```
to
```#if defined(_WIN32)
# define UA_FLOAT_IEEE754 1
#elif defined(__i386__) || defined(__x86_64__) || defined(__amd64__) || \
    defined(__ia64__) || defined(__powerpc__) || defined(__sparc__) || \
    defined(__arm__)
# define UA_FLOAT_IEEE754 1
#elif defined(__STDC_IEC_559__)
# define UA_FLOAT_IEEE754 1
#endif
#ifndef UA_FLOAT_IEEE754
# define UA_FLOAT_IEEE754 0
#endif```

jpfr, 2024-02-04:
That change does nothing. It is the same as the #elif from before.
Is there a way to detect the instruction architecture of the esp32?

m-obi, 2024-02-07:
It does change for me. Before he was redefining UA_FLOAT_IEEE754 with 0 after i defined it with 1 in my top-level CMakeLists.

jpfr, 2024-02-15:
Ah I see.
To make that explicit, please wrap with an additional #if like this.

```c
#ifndef UA_FLOAT_IEEE754
# if  defined(_WIN32)
....
# elif defined(...)
# endif
#endif
```

And add a comment that the flag can be set from the outside also (eg. with a command line flags).

Are you in a position to commit a PR?