open62541/open62541 #6261: PubSub: freed memory is accessed during writerGroup removal

MichalTyndel, 2024-02-01:
## Description

Valgrind (memcheck) detects accessing freed memory when removing writerGroups. Looks like the delayed callback, that is supposed to remove the writerGroup, is called when the structure is already deleted.  
Problem occurs when the PubSub elements are removed "bottom-up" i.e. dsw, then wg, finally connection.

## Background Information / Reproduction Steps

[example1.txt](https://github.com/open62541/open62541/files/14124627/example1.txt)
Attached is example code (c file stored as txt), that can be used to reproduce the issue. It's a mock application that establishes pubSub communication locally, then removes all PubSub objects:
- Create DataSetWriter -> DataSetReader pair that exchanges a single variable using UDP over localhost  
- Disable communication
- Remove all pubSub objects in order DSR, RG, DSW, WG, CONN, PDS.

Memcheck detects multiple invalid reads when running this example program:
[image]

Full Memcheck log is attached.
[wgRemoval.log](https://github.com/open62541/open62541/files/14124628/wgRemoval.log)

## Checklist

 - [x] open62541 Version (release number or git tag): v1.4.0-rc2-162-gcc806cb47
 - [x] Self-contained code example attached

JohannesKauffmann, 2024-02-08:
I've checked out open62541 at `cc806cb47`, built the library with `-DUA_ENABLE_AMALGAMATION=ON`, changed `#include open62541/server.h` to just `#include open62541.h`, but can't compile your example (also without amalgamation):
```
$ gcc -O2 -Wall -Wextra -fsanitize=address example1.c -I /usr/local/include/  -L /usr/local/lib/ -lopen62541.a
example1.c: In function ‘addInt32Variable’:
example1.c:28:53: error: ‘UA_ACCESSLEVELTYPE_CURRENTREAD’ undeclared (first use in this function)
   28 |                                      .accessLevel = UA_ACCESSLEVELTYPE_CURRENTREAD | UA_ACCESSLEVELTYPE_CURRENTWRITE};
      |                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
example1.c:28:53: note: each undeclared identifier is reported only once for each function it appears in
example1.c:28:86: error: ‘UA_ACCESSLEVELTYPE_CURRENTWRITE’ undeclared (first use in this function)
   28 |                                      .accessLevel = UA_ACCESSLEVELTYPE_CURRENTREAD | UA_ACCESSLEVELTYPE_CURRENTWRITE};
      |                                                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
example1.c: In function ‘addConnection’:
example1.c:37:5: error: unknown type name ‘UA_PubSubConnectionConfig’; did you mean ‘UA_ConnectionConfig’?
   37 |     UA_PubSubConnectionConfig cfg = {0};
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
      |     UA_ConnectionConfig
example1.c:39:5: error: unknown type name ‘UA_NetworkAddressUrlDataType’
   39 |     UA_NetworkAddressUrlDataType addrVal = {.url = UA_STRING((char *)((uintptr_t)addr))};
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
example1.c:39:45: error: field name not in record or union initializer
   39 |     UA_NetworkAddressUrlDataType addrVal = {.url = UA_STRING((char *)((uintptr_t)addr))};
      |                                             ^
example1.c:39:45: note: (near initialization for ‘addrVal’)
example1.c:39:52: error: incompatible types when initializing type ‘int’ using type ‘UA_String’
   39 |     UA_NetworkAddressUrlDataType addrVal = {.url = UA_STRING((char *)((uintptr_t)addr))};
      |                                                    ^~~~~~~~~
example1.c:40:30: error: request for member ‘address’ in something not a structure or union
   40 |     UA_Variant_setScalar(&cfg.address, &addrVal, &UA_TYPES[UA_TYPES_NETWORKADDRESSURLDATATYPE]);
      |                              ^
example1.c:40:60: error: ‘UA_TYPES_NETWORKADDRESSURLDATATYPE’ undeclared (first use in this function); did you mean ‘UA_NS0ID_NETWORKADDRESSURLDATATYPE’?
   40 |     UA_Variant_setScalar(&cfg.address, &addrVal, &UA_TYPES[UA_TYPES_NETWORKADDRESSURLDATATYPE]);
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                            UA_NS0ID_NETWORKADDRESSURLDATATYPE
example1.c:42:8: error: request for member ‘name’ in something not a structure or union
   42 |     cfg.name = UA_STRING((char *)((uintptr_t)name));
      |        ^
example1.c:43:8: error: request for member ‘publisherId’ in something not a structure or union
   43 |     cfg.publisherId.uint64 = pubId;
      |        ^
example1.c:44:8: error: request for member ‘publisherIdType’ in something not a structure or union
   44 |     cfg.publisherIdType = UA_PUBLISHERIDTYPE_UINT64;
      |        ^
example1.c:44:27: error: ‘UA_PUBLISHERIDTYPE_UINT64’ undeclared (first use in this function)
   44 |     cfg.publisherIdType = UA_PUBLISHERIDTYPE_UINT64;
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~
example1.c:45:8: error: request for member ‘transportProfileUri’ in something not a structure or union
   45 |     cfg.transportProfileUri = UA_STRING("http://opcfoundation.org/UA-Profile/Transport/pubsub-udp-uadp");
      |        ^
example1.c:48:25: warning: implicit declaration of function ‘UA_Server_addPubSubConnection’; did you mean ‘UA_Server_addReverseConnect’? [-Wimplicit-function-declaration]
   48 |     UA_StatusCode sts = UA_Server_addPubSubConnection(srv, &cfg, &ret);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         UA_Server_addReverseConnect
example1.c: In function ‘addWriterGroup’:
example1.c:55:5: error: unknown type name ‘UA_WriterGroupConfig’
   55 |     UA_WriterGroupConfig cfg = {0};
      |     ^~~~~~~~~~~~~~~~~~~~
example1.c:56:8: error: request for member ‘name’ in something not a structure or union
   56 |     cfg.name = UA_STRING((char *)((uintptr_t)name));
      |        ^
example1.c:57:8: error: request for member ‘writerGroupId’ in something not a structure or union
   57 |     cfg.writerGroupId = wgId;
      |        ^
example1.c:58:8: error: request for member ‘enabled’ in something not a structure or union
   58 |     cfg.enabled = true;
      |        ^
example1.c:59:8: error: request for member ‘keepAliveTime’ in something not a structure or union
   59 |     cfg.keepAliveTime = 10;
      |        ^
example1.c:60:8: error: request for member ‘publishingInterval’ in something not a structure or union
   60 |     cfg.publishingInterval = 10;
[... 245 lines cut ...]
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
example1.c:209:11: warning: implicit declaration of function ‘UA_Server_enableDataSetWriter’ [-Wimplicit-function-declaration]
  209 |     sts = UA_Server_enableDataSetWriter(srv, dswId);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
example1.c:235:5: warning: implicit declaration of function ‘UA_Server_removePubSubConnection’; did you mean ‘UA_Server_removeReverseConnect’? [-Wimplicit-function-declaration]
  235 |     UA_Server_removePubSubConnection(srv, connId);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |     UA_Server_removeReverseConnect
example1.c:236:5: warning: implicit declaration of function ‘UA_Server_removePublishedDataSet’ [-Wimplicit-function-declaration]
  236 |     UA_Server_removePublishedDataSet(srv, pdsId);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

MichalTyndel, 2024-02-13:
@JohannesKauffmann sorry for late reply. Those defines should be visible in the example. They are deifined in types_generated.h, which is included by types.h, which in turn is included by server.h.
Anyway here is CMakeLists file I used to build the example:
[CMakeLists.txt](https://github.com/open62541/open62541/files/14268340/CMakeLists.txt)

jpfr, 2024-02-16:
The fix arrives in #6284

closed 2024-02-20 by jpfr