open62541/open62541 #6269: `UA_Client_MonitoredItems_modify_async` should modify internal state?

lukasberbuer, 2024-02-10:
The async function to modify monitored items `UA_Client_MonitoredItems_modify_async` is just an inline function sending the request:

```c
static UA_INLINE UA_StatusCode
UA_Client_MonitoredItems_modify_async(UA_Client *client,
    const UA_ModifyMonitoredItemsRequest request,
    UA_ClientAsyncServiceCallback callback,
    void *userdata, UA_UInt32 *requestId) {
    return __UA_Client_AsyncService(client, &request,
        &UA_TYPES[UA_TYPES_MODIFYMONITOREDITEMSREQUEST], callback,
        &UA_TYPES[UA_TYPES_MODIFYMONITOREDITEMSRESPONSE],
        userdata, requestId);
}
```

The sync version `UA_Client_MonitoredItems_modify` on the other hand modifies the internal state: https://github.com/open62541/open62541/blob/v1.3.9/src/client/ua_client_subscriptions.c#L825-L861

So I guess, the async version should do the same? Or am I missing something?

jpfr, 2024-02-15:
Yes, you are right.
They should do the same and ideally also use identical code paths.

jpfr, 2024-02-15:
Are you in a position where you can create a pull request?

lukasberbuer, 2024-02-20:
I just created a PR #6297.