open62541/open62541 #6273: Unable to Retrieve History Data with Historicaldata Tutorial Example

iago-byte, 2024-02-13:
## Description
Greetings team,

I'm interested in using historical data functionality in my OPC UA server, and I've attempted to compile and run the "tutorial_server_historicaldata.c" example using version 1.4-rc2. However, upon reviewing the UAExpert log, I encountered the following error:
[image]

Additionally, I've checked the server log, but there is no information regarding this issue. I've also tried performing the same procedure with versions 1.3.9, 1.3.8, and 1.3.7, and in these versions, the example code seems to work without any issues. 

[image]

Could it be that I'm making a mistake in the configuration or in the usage of version 1.4-rc2?

I appreciate any guidance or advice you can provide to help resolve this issue.

Thank you!
 

## Background Information / Reproduction Steps

Used CMake options:

```bash
cmake -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DUA_NAMESPACE_ZERO=FULL -DUA_ENABLE_HISTORIZING=ON -DUA_BUILD_EXAMPLES=ON ..
```

## Checklist

Please provide the following information:

 - [x] open62541 Version: v1.3.9 and v1.4.0-rc2
 - [x] Other OPC UA SDKs used (client or server): UAexpert
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

lwmake, 2024-11-08:
I also have this problem with open62541 v1.4.6, but v1.3.14 works normally.

MicroKoder, 2025-04-09:
in the file ua_client_highlevel.c

change line:
-    item.dataEncoding = UA_QUALIFIEDNAME(0, "");
+    item.dataEncoding = UA_QUALIFIEDNAME(0, NULL);

ffreality, 2025-04-09:
> in the file ua_client_highlevel.c
> 
> change line: item.dataEncoding = UA_QUALIFIEDNAME(0, ""); to: item.dataEncoding = UA_QUALIFIEDNAME(0, NULL);

yes, figured it out, solved and actually sent a commit about it.
https://github.com/open62541/open62541/commit/5b8e113515cd6361f967253bd56dd6b835fb2d3c

At first they accepted and added it but latest version has same problem.

Noface07, 2025-05-30:
when I commented out 
```
        if (!setting->historizingBackend.timestampsToReturnSupported(
                    server,
                    setting->historizingBackend.context,
                    sessionId,
                    sessionContext,
                    &nodesToRead[i].nodeId,
                    timestampsToReturn)) {
            response->results[i].statusCode = UA_STATUSCODE_BADTIMESTAMPNOTSUPPORTED;
            continue;
        }
```

in `ua_history_database_default.c` I am getting timestamp as;

[image]

but when I monitor it in Data Access View I am getting correct source and server timestamps:

[image]

Noface07, 2025-05-31:
for a quick fix I have edited the `ua_services_attribute.c` :

added these 2 lines at 1410 and 1411:

adjustedValue.hasSourceTimestamp = true;
adjustedValue.sourceTimestamp = UA_DateTime_now();

[image]

lwmake, 2025-07-04:
Thanks

***@***.***
 
From: segotdko
Date: 2025-07-03 13:58
To: open62541/open62541
CC: lwmake; Comment
Subject: Re: [open62541/open62541] Unable to Retrieve History Data with Historicaldata Tutorial Example (Issue #6273)
segotdko left a comment (open62541/open62541#6273)
for a quick fix I have edited the ua_services_attribute.c :
added these 2 lines at 1410 and 1411:
adjustedValue.hasSourceTimestamp = true; adjustedValue.sourceTimestamp = UA_DateTime_now();
It should be enough to make your variable dynamic UA_Server_setVariableNodeDynamic(server, your node, true). This is valid for v1.4.11.